RobotLocomotion/drake #966: SharedDataHandle with no delete function leaks memory

psiorx, 2015-04-07:
It should probably be mandatory to specify a delete function when wrapping a heap allocated object in a SharedDataHandle.

Otherwise, any pointer wrapped in the handle will essentially be leaked memory.  This can be verified with many of the controller mex files.

Example:

```matlab
    while true 
      handle = SharedDataHandle(pelvisMotionControlmex(0, r.mex_model_ptr.ptr, 0, 0, zeros(6,1), zeros(6,1))); 
    end
```

This will consume all available memory until MATLAB is restarted.

@rdeits Do you agree that we need a delete function for these?

RussTedrake, 2015-04-12:
i think that the bug here is that pelvisMotionControlmex (and the other related new mex files in the controllers dir) should be using constructDrakeMexPtr , and declaring a delete function.  @rdeits -- was there a reason to not do this?  performance?  
I don't actually see any mechanism to free the ptrs that are being allocated and returned?  (maybe I missed them?)

this could even be a source of possible segfaults.  one of these mex files allocates the pointer, and that pointer is being used elsewhere (e.g. in the QP controller)... it's possible that the matlab garbage collector could try to unload the pelvisMotionControlmex from memory, thereby invalidating the pointer, and the other methods wouldn't be any the wiser.  the drake mex ptr stuff adds the locking and unlocking logic to protect against this.

SharedDataHandle is a different issue -- it's a general tool to make persistent memory.  I don't think it's behavior is in any way at fault in your example above.

rdeits, 2015-04-12:
`pelvisMotionControlmex` is part of the old controller and is not used anymore. The only mex functions used in the planeval/control loop are `threadedControllermex` which never exits and thus never has to worry about cleanup and `encodeQPInputLCMmex` which doesn't return a mex pointer. 

All those old mex files are on my list to zap with the next pull request, as soon as I convert the remaining drake Atlas examples to the new controller.

RussTedrake, 2015-04-12:
Excellent.  Sounds like a plan.  Thanks.

closed 2015-04-20 by RussTedrake