RobotLocomotion/drake #967: Error in transverseLQR

xParas, 2015-04-07:
Hi,

 I am running transverseLQR from the compass gait example. I have installed the required SNOPT and its lic files. I am getting the following error. I am on windows and matlab R2013a

 runTransverseCycle
Elapsed time is 12.124595 seconds.
Error using Trajectory/fnplt (line 323)
plotdims out of range

Error in HybridTrajectory/fnplt (line 98)
        hn=fnplt(obj.traj{i},plotdims);

Error in runTransverseCycle (line 9)
fnplt(xtraj,[2 4]);

screen shot added.

[image]

Thanks,
-Paras

RussTedrake, 2015-04-10:
I got the same results on my system.  The `dev` subdirectories are not maintained directly -- they are not part of the unit test suite so can get out of date.  

i've taken a quick pass at updating this example.  getting the transverseLQR working again was easy.  i got a lot of the way through making the transverse verification work again (but have to run to a meeting now!).  will put in my fixes in a pull request.

- Russ

xParas, 2015-04-10:
OK thanks for the update, will look forward to get it working at my end. On a larger part what I want is to study drake and try implement the oscillators I am working on with compass gait in drake and run the stability analysis you do in your examples. 
I am still on learning curve with drake, but guess when I am done, it will be good results with my work. 
Thanks.

RussTedrake, 2015-08-27:
closing this for now.  would be happy to see any updates if you have them.

closed 2015-08-27 by RussTedrake