RobotLocomotion/drake #974: Eigen as a required dependency?

tkoolen, 2015-04-10:
I'd like to propose making Eigen a required dependency. My arguments:
* As we're moving more stuff to C++, a larger percentage of our codebase will be unusable without Eigen.
* It's a hassle to remember to use if (eigen3_FOUND) everywhere.
* Why would any system we care about not have Eigen?

What do you think?

psiorx, 2015-04-10:
I agree.

avalenzu, 2015-04-10:
I'd be in favor of Eigen as a required dependency too.

RussTedrake, 2015-04-11:
I agree, too.  I will have to update the distros, but that's easy enough.

in fact, i think we should make matlab optional.  There are definitely systems where we want to be able to compile the c++ tools which don't have matlab.

andybarry, 2015-04-11:
> matlab optional

exciting!

PR #987 (2015-04-15): eigen3 is now a required dependency.  matlab is an optional dependency.
resolves #974
closed 2015-04-15 by RussTedrake