RobotLocomotion/drake #981: Error Compiling Drake on Yosemite: gfortran: warning: couldn't understand kern.osversion '14.3.0 ld: -rpath can only be used when targeting Mac OS X 10.5 or later

tarquasso, 2015-04-12:
Problems compiling drake:

[100%] Building Fortran object src/CMakeFiles/avl.dir/second_g77.f.o
gfortran: warning: couldn't understand kern.osversion '14.3.0
Warning: Nonexistent include directory "/Users/rkk/drake-distro/avl/pod-build/include"
Linking Fortran executable ../bin/avl
gfortran: warning: couldn't understand kern.osversion '14.3.0
ld: -rpath can only be used when targeting Mac OS X 10.5 or later
collect2: error: ld returned 1 exit status
make[4]: *** [bin/avl] Error 1
make[3]: *** [src/CMakeFiles/avl.dir/all] Error 2
make[2]: *** [all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

I am running Mac OS X Yosemite... To describe what I did: I was 53 commits behind on the drake/rigidbody repo, so I ran
$ git pull
Here is now my 
$git status
output:
tarqu:drake-distro rkk$ git status
# On branch rigidbody
# Changes not staged for commit:
#   (use "git add <file>..." to update what will be committed)
#   (use "git checkout -- <file>..." to discard changes in working directory)
#   (commit or discard the untracked or modified content in submodules)
#
#	modified:   avl (new commits)
#	modified:   bullet (new commits, untracked content)
#	modified:   drake (new commits, modified content, untracked content)
#	modified:   eigen (new commits, untracked content)
#	modified:   lcm (new commits, untracked content)
#	modified:   libbot (new commits, untracked content)
#	modified:   spotless (new commits, modified content)
#	modified:   xfoil (new commits)
#
# Untracked files:
#   (use "git add <file>..." to include in what will be committed)
#
#	#README#
#	.#README
no changes added to commit (use "git add" and/or "git commit -a")

Then I made sure my prerequisites are all there, X11 was missing, so I installed that one. Also followed the steps here:
https://github.com/RobotLocomotion/drake/wiki/Homebrew, and added this line:
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/opt/X11/lib/pkgconfig
to my ~/.bash_profile

I googled the make error and found http://stackoverflow.com/questions/26486163/gcc-4-9-1-in-os-x-yosemite-gcc-warning-couldn-t-understand-kern-osversion-1
My command line tools are up to date, so that does not help.

Any idea what I can do to get over this compile error? 
Thanks

RussTedrake, 2015-04-12:
> (use "git add ..." to update what will be committed)
> (use "git checkout -- ..." to discard changes in working directory)
> (use "git add ..." to include in what will be committed)

it looks like you’re only have trouble building avl (which you probably don’t actually need — it computes aerodynamics coefficients).  one work-around is to skip building avl by commenting it out in drake-distro/tobuild.txt .  

as far as actually resolving the problem… is it possible that you’ve upgraded the mac since you build avl last?  maybe try doing `make clean` and then `make VERBOSE=1` in the avl subdirectory?

tarquasso, 2015-04-12:
I commented out ‘avl'. Same problem now for running make ‘xfoil':
[100%] Building Fortran object src/CMakeFiles/xfoil.dir/ntcalc.f.o
gfortran: warning: couldn't understand kern.osversion '14.3.0
Warning: Nonexistent include directory "/Users/rkk/drake-distro/xfoil/pod-build/include"
Linking Fortran executable ../bin/xfoil
gfortran: warning: couldn't understand kern.osversion '14.3.0
ld: -rpath can only be used when targeting Mac OS X 10.5 or later
collect2: error: ld returned 1 exit status
make[4]: *** [bin/xfoil] Error 1
make[3]: *** [src/CMakeFiles/xfoil.dir/all] Error 2
make[2]: *** [all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

Here is the output from running the make VERBOSE=1 for 'avl':

tarqu:drake-distro rkk$ cd avl/
tarqu:avl rkk$ make clean
tarqu:avl rkk$ make VERBOSE=1
/Applications/Xcode.app/Contents/Developer/usr/bin/make configure

BUILD_PREFIX: /Users/rkk/drake-distro/build

# create the temporary build directory if needed
# run CMake to generate and configure the build scripts
Re-run cmake no build system arguments
-- The C compiler identification is AppleClang 6.1.0.6020049
-- The CXX compiler identification is AppleClang 6.1.0.6020049
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- The Fortran compiler identification is GNU
-- Check for working Fortran compiler: /usr/local/bin/gfortran
-- Check for working Fortran compiler: /usr/local/bin/gfortran  -- works
-- Detecting Fortran compiler ABI info
-- Detecting Fortran compiler ABI info - done
-- Checking whether /usr/local/bin/gfortran supports Fortran 90
-- Checking whether /usr/local/bin/gfortran supports Fortran 90 -- yes
-- Looking for XOpenDisplay in /usr/X11R6/lib/libX11.dylib;/usr/X11R6/lib/libXext.dylib
-- Looking for XOpenDisplay in /usr/X11R6/lib/libX11.dylib;/usr/X11R6/lib/libXext.dylib - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/X11R6/lib/libX11.dylib
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rkk/drake-distro/avl/pod-build
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C pod-build all install
/usr/local/Cellar/cmake/3.0.0/bin/cmake -H/Users/rkk/drake-distro/avl -B/Users/rkk/drake-distro/avl/pod-build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.0.0/bin/cmake -E cmake_progress_start /Users/rkk/drake-distro/avl/pod-build/CMakeFiles /Users/rkk/drake-distro/avl/pod-build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f plotlib/CMakeFiles/xwin.dir/build.make plotlib/CMakeFiles/xwin.dir/depend
cd /Users/rkk/drake-distro/avl/pod-build && /usr/local/Cellar/cmake/3.0.0/bin/cmake -E cmake_depends "Unix Makefiles" /Users/rkk/drake-distro/avl /Users/rkk/drake-distro/avl/plotlib /Users/rkk/drake-distro/avl/pod-build /Users/rkk/drake-distro/avl/pod-build/plotlib /Users/rkk/drake-distro/avl/pod-build/plotlib/CMakeFiles/xwin.dir/DependInfo.cmake --color=
Dependee "/Users/rkk/drake-distro/avl/pod-build/plotlib/CMakeFiles/xwin.dir/DependInfo.cmake" is newer than depender "/Users/rkk/drake-distro/avl/pod-build/plotlib/CMakeFiles/xwin.dir/depend.internal".
Dependee "/Users/rkk/drake-distro/avl/pod-build/plotlib/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/Users/rkk/drake-distro/avl/pod-build/plotlib/CMakeFiles/xwin.dir/depend.internal".
Scanning dependencies of target xwin
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f plotlib/CMakeFiles/xwin.dir/build.make plotlib/CMakeFiles/xwin.dir/build
/usr/local/Cellar/cmake/3.0.0/bin/cmake -E cmake_progress_report /Users/rkk/drake-distro/avl/pod-build/CMakeFiles 39
[... 520 lines cut ...]
/usr/local/Cellar/cmake/3.0.0/bin/cmake -E cmake_progress_report /Users/rkk/drake-distro/xfoil/pod-build/CMakeFiles 15
[ 38%] Building Fortran object src/CMakeFiles/xfoil.dir/xpanel.f.o
cd /Users/rkk/drake-distro/xfoil/pod-build/src && /usr/local/bin/gfortran  -DUNDERSCORE -O3 -I/Users/rkk/drake-distro/xfoil/pod-build/include -I/Users/rkk/drake-distro/build/include    -c /Users/rkk/drake-distro/xfoil/src/xpanel.f -o CMakeFiles/xfoil.dir/xpanel.f.o
gfortran: warning: couldn't understand kern.osversion '14.3.0
Warning: Nonexistent include directory "/Users/rkk/drake-distro/xfoil/pod-build/include"
/usr/local/Cellar/cmake/3.0.0/bin/cmake -E cmake_progress_report /Users/rkk/drake-distro/xfoil/pod-build/CMakeFiles 16
[ 40%] Building Fortran object src/CMakeFiles/xfoil.dir/xoper.f.o
cd /Users/rkk/drake-distro/xfoil/pod-build/src && /usr/local/bin/gfortran  -DUNDERSCORE -O3 -I/Users/rkk/drake-distro/xfoil/pod-build/include -I/Users/rkk/drake-distro/build/include    -c /Users/rkk/drake-distro/xfoil/src/xoper.f -o CMakeFiles/xfoil.dir/xoper.f.o
gfortran: warning: couldn't understand kern.osversion '14.3.0
[... cut ...]
/usr/local/Cellar/cmake/3.0.0/bin/cmake -E cmake_progress_report /Users/rkk/drake-distro/xfoil/pod-build/CMakeFiles 19
[ 47%] Building Fortran object src/CMakeFiles/xfoil.dir/xqdes.f.o
cd /Users/rkk/drake-distro/xfoil/pod-build/src && /usr/local/bin/gfortran  -DUNDERSCORE -O3 -I/Users/rkk/drake-distro/xfoil/pod-build/include -I/Users/rkk/drake-distro/build/include    -c /Users/rkk/drake-distro/xfoil/src/xqdes.f -o CMakeFiles/xfoil.dir/xqdes.f.o
gfortran: warning: couldn't understand kern.osversion '14.3.0
Warning: Non

RussTedrake, 2015-04-12:
based on a quick google, it looks like you probably forgot to install the updated command-line tools after update xcode. 

try this on your command line:
```
xcode-select --install 
```
then try again.

(fwiw - avl and xfoil are ying and yang, i should have told you to take both out of your build.)

On Apr 12, 2015, at 4:03 PM, Robert Katzschmann <notifications@github.com> wrote:

> I commented out ‘avl'. Same problem now for running make ‘xfoil':
> [100%] Building Fortran object src/CMakeFiles/xfoil.dir/ntcalc.f.o
> gfortran: warning: couldn't understand kern.osversion '14.3.0 [...]

RussTedrake, 2015-04-14:
i will close this assuming it is resolved.  if you can confirm, that would be much appreciated.  reopen the issue if it is not resolved.

closed 2015-04-14 by RussTedrake

tarquasso, 2015-04-14:
Commenting out xfoil and rerunning the make of drake gives me this error:

CMake Error at cmake/mex.cmake:274 (message):
  Your cmake C compiler is: "/usr/bin/cc" but your mex options use: "".  You
  must use the same compilers.  You can either:

    a) reconfigure the mex compiler by running 'mex -setup' in  MATLAB, or
    b) Set the default compiler for cmake by setting the CC environment variable in your terminal.

Call Stack (most recent call first):
  CMakeLists.txt:41 (include)

Matlab tells me this:

>> mex -setup C
MEX configured to use 'Xcode with Clang' for C language compilation.
Warning: The MATLAB C and Fortran API has changed to support MATLAB
	 variables with more than 2^32-1 elements. In the near future
	 you will be required to update your code to utilize the
	 new API. You can find more information about this at:
	 http://www.mathworks.com/help/matlab/matlab_external/upgrading-mex-files-to-use-64-bit-api.html <http://www.mathworks.com/help/matlab/matlab_external/upgrading-mex-files-to-use-64-bit-api.html>.

What can I do to get the mex options adjusted? Where in CMakeLists shall i make an adjustment?

I regards to avl and xfoil:
I ran:
$ xcode-select --install 
xcode-select: error: command line tools are already installed, use "Software Update" to install updates
I made sure xcode software updates are all installed, still it does not compile and ends with the same error. Any ideas?

RussTedrake, 2015-04-14:
the mex -setup line output looks good.  but if you have changed it, you will have to rerun make configure in the offending directory.

not sure about the gfortran issue.  we have other machine running yosemite and compiling it aok.  it would definitely be good to figure this one out.

tarquasso, 2015-04-14:
I haven’t changed the setup of the mex, I just ran mex -setup and got the output I pasted into my previous email. It still gives me that error. 
running $make configure in the drake subdirectory gives me still errors:

-- found mex: /Applications/MATLAB_R2014b.app/bin/mex
CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CC using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:97 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:99 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CDEBUGFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:100 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_COPTIMFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:101 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CLIBS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:102 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CXX using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:105 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CXXDEBUGFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:106 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CXXOPTIMFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:107 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_CXXLIBS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:108 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_LD using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:119 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_LDFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:120 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_LDDEBUGFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:121 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_LDOPTIMFLAGS using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:122 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Warning at cmake/mex.cmake:18 (message):
  Could not find MEX_LDEXTENSION using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:123 (get_mex_option)
  cmake/mex.cmake:270 (mex_setup)
  CMakeLists.txt:41 (include)

CMake Error at cmake/mex.cmake:257 (get_compiler_version):
  get_compiler_version Function invoked with incorrect arguments for function
  named: get_compiler_version
Call Stack (most recent call first):
  cmake/mex.cmake:272 (compare_compilers)
  CMakeLists.txt:41 (include)

CMake Error at cmake/mex.cmake:274 (message):
  Your cmake C compiler is: "/usr/bin/cc" but your mex options use: "".  You
  must use the same compilers.  You can either:

    a) reconfigure the mex compiler by running 'mex -setup' in  MATLAB, or
    b) Set the default compiler for cmake by setting the CC environment variable in your terminal.

Call Stack (most recent call first):
  CMakeLists.txt:41 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/rkk/drake-distro/drake/pod-build/CMakeFiles/CMakeOutput.log".
make: *** [configure] Error 1
tarqu:drake rkk$ 

I am not sure how I should do this here to fix the problem:

> Your cmake C compiler is: "/usr/bin/cc" but your mex options use: "". You
> must use the same compilers. You can either:
> 
> a) reconfigure the mex compiler by running 'mex -setup' in MATLAB, or
> b) Set the default compiler for cmake by setting the CC environment variable in your terminal.

RussTedrake, 2015-04-14:
can you run `/Applications/MATLAB_R2014b.app/bin/mex -v pod-build/dummy.c` at the command line and paste the output?

tarquasso, 2015-04-14:
tarqu:drake rkk$ /Applications/MATLAB_R2014b.app/bin/mex -v pod-build/dummy.c
Verbose mode is on.
/Users/rkk/drake-distro/drake/pod-build/dummy.c not found; check that you are in the correct current folder, and check the spelling of '/Users/rkk/drake-distro/drake/pod-build/dummy.c'.
tarqu:drake rkk$ cd pod-build
tarqu:pod-build rkk$ ls
CMakeCache.txt		CMakeFiles		CTestCustom.cmake

dont have dummy.c…

RussTedrake, 2015-04-14:
it gets created during the build with a simple `touch pod-build/dummy.c` . you can do it manually

tarquasso, 2015-04-14:
tarqu:drake rkk$ /Applications/MATLAB_R2014b.app/bin/mex -v pod-build/dummy.c
Verbose mode is on.
Neither -compatibleArrayDims nor -largeArrayDims is selected.
	 Using -compatibleArrayDims. In the future, MATLAB will require the use of
	 -largeArrayDims and remove the -compatibleArrayDims option.
	 For more information:
	 http://www.mathworks.com/help/matlab/matlab_external/upgrading-mex-files-to-use-64-bit-api.html.
... Looking for compiler 'Xcode with Clang' ...
... Looking for environment variable 'DEVELOPER_DIR' ...No.
... Executing command 'xcode-select -print-path' ...Yes ('/Applications/Xcode.app/Contents/Developer').
... Looking for folder '/Applications/Xcode.app/Contents/Developer' ...Yes.
... Executing command 'which xcrun' ...Yes ('/usr/bin/xcrun').
... Looking for folder '/usr/bin' ...Yes.
... Executing command 'defaults read com.apple.dt.Xcode IDEXcodeVersionForAgreedToGMLicense' ...No.
... Executing command 'defaults read /Library/Preferences/com.apple.dt.Xcode IDEXcodeVersionForAgreedToGMLicense' ...Yes ('6.1').
... Executing command '
agreed=6.1 
 if echo $agreed | grep -E '[\.\"]' >/dev/null; then 
 lhs=`expr "$agreed" : '\([0-9]*\)[\.].*'` 
  rhs=`expr "$agreed" : '[0-9]*[\.]\(.*\)$'` 
 if echo $rhs | grep -E '[\."]' >/dev/null; then 
 rhs=`expr "$rhs" : '\([0-9]*\)[\.].*'` 
 fi 
 if [ $lhs -gt 4 ] || ( [ $lhs -eq 4 ] && [ $rhs -ge 3 ] ); then 
 echo $agreed 
 else 
 exit 1
 fi 
 fi' ...Yes ('6.1').
... Executing command 'xcode-select -print-path' ...Yes ('/Applications/Xcode.app/Contents/Developer').
... Looking for folder '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.7.sdk' ...No.
... Looking for folder '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.8.sdk' ...No.
... Looking for folder '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk' ...Yes.
... Executing command 'xcode-select -print-path' ...Yes ('/Applications/Xcode.app/Contents/Developer').
... Looking for folder '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.7.sdk' ...No.
... Looking for folder '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.8.sdk' ...No.
... Looking for folder '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk' ...Yes.
... Executing command 'echo /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk | rev | cut -c1-10 | rev | egrep -oh '[0-9]+\.[0-9]+'' ...Yes ('10.9').
Found installed compiler 'Xcode with Clang'.
Options file details
-------------------------------------------------------------------
	Compiler location: /Applications/Xcode.app/Contents/Developer
	Options file: /Users/rkk/.matlab/R2014b/mex_C_maci64.xml
	CMDLINE100 : /usr/bin/xcrun -sdk macosx10.9 clang -c -DMX_COMPAT_32   -DMATLAB_MEX_FILE -I"/Applications/MATLAB_R2014b.app/extern/include" -I"/Applications/MATLAB_R2014b.app/simulink/include" -fno-common -arch x86_64 -mmacosx-version-min=10.9 -fexceptions -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk -O2 -DNDEBUG /Users/rkk/drake-distro/drake/pod-build/dummy.c -o /var/folders/hp/42hp13353gs4cv4hl78gj4zw0000gn/T//mex_1096130455862_20487/dummy.o
	CMDLINE200 : /usr/bin/xcrun -sdk macosx10.9 clang -Wl,-twolevel_namespace -undefined error -arch x86_64 -mmacosx-version-min=10.9 -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk -bundle  -Wl,-exported_symbols_list,"/Applications/MATLAB_R2014b.app/extern/lib/maci64/mexFunction.map" /var/folders/hp/42hp13353gs4cv4hl78gj4zw0000gn/T//mex_1096130455862_20487/dummy.o  -O -Wl,-exported_symbols_list,"/Applications/MATLAB_R2014b.app/extern/lib/maci64/mexFunction.map"  -L"/Applications/MATLAB_R2014b.app/bin/maci64" -lmx -lmex -lmat -lstdc++ -o dummy.mexmaci64
	CMDLINE300 : rm -f /var/folders/hp/42hp13353gs4cv4hl78gj4zw0000gn/T//mex_1096130455862_20487/dummy.o
	CC : /usr/bin/xcrun -sdk macosx10.9 clang
	DEFINES : -DMX_COMPAT_32   -DMATLAB_MEX_FILE
	MATLABMEX : -DMATLAB_MEX_FILE
	CFLAGS : -fno-common -arch x86_64 -mmacosx-version-min=10.9 -fexceptions -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk
	INCLUDE : -I"/Applications/MATLAB_R2014b.app/extern/include" -I"/Applications/MATLAB_R2014b.app/simulink/include"
	COPTIMFLAGS : -O2 -DNDEBUG
	CDEBUGFLAGS : -g
	LD : /usr/bin/xcrun -sdk macosx10.9 clang
	LDFLAGS : -Wl,-twolevel_namespace -undefined error -arch x86_64 -mmacosx-version-min=10.9 -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk -bundle  -Wl,-exported_symbols_list,"/Applications/MATLAB_R2014b.app/extern/lib/maci64/mexFunction.map"
	LDBUNDLE : -bundle 
	LINKEXPORT : -Wl,-exported_symbols_list,"/Applications/MATLAB_R2014b.app/extern/lib/maci64/mexFunction.map"
	LINKLIBS : -L"/Applications/MATLAB_R2014b.app/bin/maci64" -lmx -lmex -lmat -lstdc++
	LDOPTIMFLAGS : -O
	LDDEBUGFLAGS : -g
	OBJEXT : .o
	LDEXT : .mexmaci64
	SETENV : CC="/usr/bin/xcrun -sdk macosx10.9 clang"
CXX="/usr/bin/xcrun -sdk macosx10.9 clang"
[... cut ...] v4hl78gj4zw0000gn/T//mex_1096130455862_20487/dummy.o  -O -Wl,-exported_symbols_list,"/Applications/MATLAB_R2014b.app/extern/lib/maci64/mexFunction.map"  -L"/Applications/MATLAB_R2014b.app/bin/maci64" -lmx -lmex -lmat -lstdc++ -o dummy.mexmaci64
Undefined symbols for architecture x86_64:
  "_mexFunction", referenced from:
     -exported_symbol[s_list] command line option
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

RussTedrake, 2015-04-14:
that all looks fine to me.  

we'd have to step in and figure out why this line is failing:

> > -- found mex: /Applications/MATLAB_R2014b.app/bin/mex
> > CMake Warning at cmake/mex.cmake:18 (message):
> > Could not find MEX_CC using mex -v
> > Call Stack (most recent call first):
> > cmake/mex.cmake:97 (get_mex_option)
> > cmake/mex.cmake:270 (mex_setup)
> > CMakeLists.txt:41 (include)
> >

tarquasso, 2015-04-14:
I am in Italy, back on Monday at MIT, lets do it then in person or can I do any more debugging from here?

tarquasso, 2015-04-18:
Good News: I started off from scratch by cloning drake [rigidbody] again into a new folder, running all of the steps in the installation guide and got it all to compile without commenting out any of the submodules.

I think the problem might have been that the gcc dependency for avl was not correcty installed and also not correctly linked (because of gcc49 being linked).
$brew unlink gcc49
$brew unlink gfortran

and the rerun
$brew link gcc

fixed that. See details below:

UPDATING HOMEBREW PREREQUESITES:
drake-distro rkk$ brew update
(…)
installing prereqs for avl
==> Installing dependencies for gcc: gmp, mpfr, libmpc, isl
==> Installing gcc dependency: gmp
==> Downloading https://homebrew.bintray.com/bottles/gmp-6.0.0a.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring gmp-6.0.0a.yosemite.bottle.tar.gz
🍺  /usr/local/Cellar/gmp/6.0.0a: 15 files, 3.2M
==> Installing gcc dependency: mpfr
==> Downloading https://homebrew.bintray.com/bottles/mpfr-3.1.2-p11.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring mpfr-3.1.2-p11.yosemite.bottle.tar.gz
🍺  /usr/local/Cellar/mpfr/3.1.2-p11: 24 files, 3.5M
==> Installing gcc dependency: libmpc
==> Downloading https://homebrew.bintray.com/bottles/libmpc-1.0.3.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libmpc-1.0.3.yosemite.bottle.tar.gz
🍺  /usr/local/Cellar/libmpc/1.0.3: 10 files, 380K
==> Installing gcc dependency: isl
==> Downloading https://homebrew.bintray.com/bottles/isl-0.12.2.yosemite.bottle.2.tar.gz
######################################################################## 100.0%
==> Pouring isl-0.12.2.yosemite.bottle.2.tar.gz
🍺  /usr/local/Cellar/isl/0.12.2: 55 files, 3.1M
==> Installing gcc
==> Downloading https://homebrew.bintray.com/bottles/gcc-4.9.2_1.yosemite.bottle.1.tar.gz
######################################################################## 100.0%
==> Pouring gcc-4.9.2_1.yosemite.bottle.1.tar.gz
==> Caveats
GCC has been built with multilib support. Notably, OpenMP may not work:
  https://gcc.gnu.org/bugzilla/show_bug.cgi?id=60670
If you need OpenMP support you may want to
  brew reinstall gcc --without-multilib
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink bin/c++-4.9
Target /usr/local/bin/c++-4.9
is a symlink belonging to gcc49. You can unlink it:
  brew unlink gcc49

To force the link and overwrite all conflicting files:
  brew link --overwrite gcc

To list all files that would be deleted:
  brew link --overwrite --dry-run gcc

Possible conflicting files are:
(…)
==> Summary
🍺  /usr/local/Cellar/gcc/4.9.2_1: 1156 files, 203M
installing prereqs for xfoil
Warning: gcc-4.9.2_1 already installed, it's just not linked
installing prereqs for drake
java version >= 1.6: YES
Error: boost-1.55.0 already installed
To install this version, first `brew unlink boost'
Error: graphviz-2.36.0 already installed
To install this version, first `brew unlink graphviz'
tarqu:drake-distro rkk$ ma

FIX BREW LINK OF GCC:

tarqu:drake-distro rkk$ brew unlink gcc49
Unlinking /usr/local/Cellar/gcc49/4.9.2_1... 25 symlinks removed
tarqu:drake-distro rkk$ brew link gcc
Linking /usr/local/Cellar/gcc/4.9.2_1... 
Error: Could not symlink bin/gfortran
Target /usr/local/bin/gfortran
is a symlink belonging to gfortran. You can unlink it:
  brew unlink gfortran

To force the link and overwrite all conflicting files:
  brew link --overwrite gcc

To list all files that would be deleted:
  brew link --overwrite --dry-run gcc
tarqu:drake-distro rkk$ brew unlink gfortran
Unlinking /usr/local/Cellar/gfortran/4.8.2... 3 symlinks removed
tarqu:drake-distro rkk$ brew link gcc
Linking /usr/local/Cellar/gcc/4.9.2_1... 29 symlinks created
tarqu:drake-distro rkk$ 

RERUNNING HOMEBREW PREREQS INSTALL:
tarqu:drake-distro rkk$ ./install_prereqs.sh homebrew
Warning: pkg-config-0.28 already installed
Error: cmake-3.0.0 already installed
To install this version, first `brew unlink cmake'
Error: gtk+-2.24.24 already installed
To install this version, first `brew unlink gtk+'
installing prereqs for lcm
Error: wget-1.15 already installed
To install this version, first `brew unlink wget'
Error: xz-5.0.5 already installed
To install this version, first `brew unlink xz'
Error: glib-2.40.0_1 already installed
To install this version, first `brew unlink glib'
Error: coreutils-8.22 already installed
To install this version, first `brew unlink coreutils'
installing prereqs for libbot
Warning: jpeg-8d already installed
Error: cmake-3.0.0 already installed
To install this version, first `brew unlink cmake'
Error: gtk+-2.24.24 already installed
To install this version, first `brew unlink gtk+'
Error: libpng-1.6.12 already installed
To install this version, first `brew unlink libpng'
installing prereqs for eigen
installing prereqs for bullet
Warning: gnu-sed-4.2.2 already installed
installing prereqs for avl
Warning: gcc-4.9.2_1 already installed
installing prereqs for xfoil
Warning: gcc-4.9.2_1 already installed
installing prereqs for drake
java version >= 1.6: YES
Error: boost-1.55.0 already installed
To install this version, first `brew unlink boost'
Error: graphviz-2.36.0 already installed
To install this version, first `brew unlink graphviz'

I am now running 
$make test 
in the drake-distro folder, that seems to be outputting some errors, but will open another issue ticket for that.

 
On Apr 14, 2015, at 20:25, Robert Katzschmann <robert@katzschmann.de> wrote:

I am in Italy, back on Monday at MIT, lets do it then in person or can I do any more debugging from here?

RussTedrake, 2015-04-18:
Excellent.  Thanks for posting the resolution

tkoolen, 2016-04-11:
I had similar errors (in xfoil, but also in avl). For me, removing drake-distro/pod-build/avl-prefix and drake-distro/pod-build/download-avl-prefix and rerunning make avl fixed avl. xfoil got fixed after doing the unlink/link thing for gcc, but I'm betting that I could have also just deleted the prefix folders for xfoil.