RobotLocomotion/drake #982: Error running Atlas simulation on master

rdeits, 2015-04-13:
I'm getting this error after a clean build of Drake master:

```
Error using solveLCPmex
Function "mxGetIr" is obsolete.
(64-bit mex files using sparse matrices must be rebuilt with the "-largeArrayDims" option.  See the R2006b release notes for more details.)

Error in TimeSteppingRigidBodyManipulator/solveMexLCP (line 250)
        [z, Mqdn, wqdn] = solveLCPmex(obj.manip.mex_model_ptr, q, v, u, ph n, D, obj.timestep, obj.z_inactive_guess_tol, obj.LCP_cache.data.z, H, C,;

Error in TimeSteppingRigidBodyManipulator/update (line 196)
           [obj,z,Mqdn,wqdn] = solveMexLCP(obj,t,x,u);

Error in FeedbackSystem/update (line 67)
      if (obj.sys1.getNumDiscStates()) xdn=[xdn;update(obj.sys1,t,x1,sat1(,y2+u))]; end

Error in DynamicalSystem/simulate (line 110)
simout = sim(mdl,pstruct);

Error in runAtlasPelvisDrop (line 94)
ytraj = simulate(sys, [0, T], x0, struct('gui_control_interface', true));
Error using DynamicalSystem/simulate (line 110)
Error reported by S-function 'DCSFunction' in
'MIMOCascade_3237125/system1/DrakeSys':

DrakeSystem S-Function: error MATLAB:err_sparse_64bit_obsolete in MATLAB
callback.
See additional debugging information above

Error in runAtlasPelvisDrop (line 94)
ytraj = simulate(sys, [0, T], x0, struct('gui_control_interface', true));
```

The call to `mxGetIr` seems to come from `drakeUtil.cpp`'s `eigenToMatlabSparse` function.

rdeits, 2015-04-13:
(this is on my DRC laptop with R2014a on linux)

RussTedrake, 2015-04-13:
the -largeArrayDims option should be enabled in the newest cmake.  perhaps you need to `git submodule update --init` in drake?

rdeits, 2015-04-13:
Yup, looks like my cmake was out of date

closed 2015-04-13 by rdeits