RobotLocomotion/drake #997: Matlab crash involving TaylorVar and a sparse matrix

tkoolen, 2015-04-16:
Got stuck on this one for a while. Not sure if this is a Drake issue or a Matlab issue.

Minimal failing test:
```
function matlabCrash()
n = 2; % crash for n > 1
nq = 1; % crash for any n > 0
f = randn(n, 1);
df = {randn(n, nq)};
a = TaylorVar(f, df);

full_matrix = zeros(n, 1);
full_matrix(:, 1) = a; % warning, but no crash

sparse_matrix = sparse(n, 1);
sparse_matrix(:, 1) = a; % crash
end
```

Crashed for me on OSX with Matlab 2014a. Could you guys run this on your machine and let me know whether it crashed or not?

RussTedrake, 2015-04-16:
not surprising, but i get the same behavior on my mac running 2014b.

RussTedrake, 2015-04-16:
interestingly, 

```
sparse_matrix(:, 1) = a; % crash
```
definitely calls `TaylorVar.double()` 

but changing the line to 
```
sparse_matrix(:,1) = double(a)
```
resolves the crash.

pretty suspicious.

psiorx, 2015-04-16:
Crashes for me on R2014a @ Ubuntu 14.04

tkoolen, 2015-04-16:
> changing the line to
> ```
> sparse_matrix(:,1) = double(a)
> ```
> resolves the crash.
OK, that could be our workaround then.

closed 2015-08-27 by RussTedrake