RobotLocomotion/drake #1002: Error introduced by recent change: simulateODE

KeyDamo, 2015-04-17:
The runLQR example does not originally include the ```[ytraj,xtraj,lcmlog] =``` output arguments, but they are in the definition of ```simulate``` and I was using them until this error appeared in some commit out of the last 300 or so (about 1 week).
[image]

RussTedrake, 2015-04-17:
The lcmlog output has not been implemented yet in simulateODE.  My guess is that you had it working on a system with simulink installed, and now you are seeing these differences because your matlab does not include simulink.

(in normal operation, simulateODE only gets called if simulink is not available -- admittedly we don't use it much here but will try to support it within reason)

KeyDamo, 2015-04-17:
Digging a bit deeper:
```checkDependency('simulink')``` returns 1 in both my old Drake version and the current one.
I also verified that the lines that divert to ```simulate``` or ```simulateODE``` are the same in my old Drake version and the current one (487-494 in DrakeSystem.m): 
``` matlab
    function varargout = simulate(obj,varargin)
      varargout=cell(1,max(nargout,1));
      if checkDependency('simulink') && exist('DCSFunction','file')
        [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});
      else
        [varargout{:}] = simulateODE(obj,varargin{:});
      end
    end
```
Right now I just verified that ```exist('DCSFunction','file')``` returns 0 in the current Drake while it returns 3 in my old Drake version. So this is where the problem is.

RussTedrake, 2015-04-17:
is it possible that you have not run make?

KeyDamo, 2015-04-17:
That's the compilation part right? I'm only synced with the drake folder inside the drake-precompiled-0.9.8-2fce1a1-win64 folder (you got me, I'm using win64) and have never run ```make```. Do you think something might have changed in the other folders?

RussTedrake, 2015-04-17:
you should have DCSFunction.mexw64 (I believe that's the correct extension) in your `drake/systems/` directory, and the `drake/systems` directory should be in your path.

KeyDamo, 2015-04-17:
That's the right extention according to ```mexext```.
Right. I just verified that I don't have that file in the current Drake whereas it's there in the old one. So I guess I just have to compile DCSfunction.cpp.
Well, I just copied the *.lib, *.ext and *.mexw64 from the old Drake and it worked.
I have not entered into the details of the compilation of the *.cpp files in Drake. If I encounter problems when I do I'll let you know.
Thanks!

closed 2015-04-17 by KeyDamo