RobotLocomotion/drake #1003: Strange IK trajectory

steinachim, 2015-04-17:
Hi,

I'm currently working on generating trajectories between two robot joint configurations using MATLAB `inverseKinTraj`. Now, I'd expect a pretty direct trajectory transitioning between the two configurations, but what I (consistently) receive is the following trajectory:
https://gist.github.com/steinachim/96af7881c64dde0a4c79

In dynamics planning I could understand gathering momentum for the upswing, but this should be simple inverse kinematics, right? Am I missing something? What I am looking for is a trajectory that starts at t = 0 (without the wait seen in the video) and then slowly moves the arm to the target position without deviation.

For the above example, time parametrization was `t = 0 : 0.5 : 5` and both nominal and seed trajectory were kept at the start posture to keep things simple.
As for the constraints used:
1. `WorldPositionConstraint` and `WorldOrientationConstraint` for both feet (for the total duration)
2. `QuasiStaticConstraint` during the whole motion
3. `PostureConstraint` for the arm joints at the end (t=5)

Any help would be appreciated as I'm getting quite puzzled by this behaviour.

Best,
Achim

hongkai-dai, 2015-04-17:
I think you have too many knot points (11) in total. What inverseKinTraj
does is to interpolate those knot points with cubic splines, and minimize
the sum of squares on configuration acceleration at those knot points. When
you have that many knot points, the arm can swing back and forth to achieve
a smaller cost on the total acceleration. Try use 3 or 4 knot points
instead (it will also be faster to use less knot points).

steinachim, 2015-04-17:
Thanks for the super-fast response.
I just tried that and the only way I can get rid of the swinging is if I only allow two time-points (`t = [0, 5]`). Adding even one more leads to the same problem. 
And am I correct to assume that the constraints will only be checked at those time-points (if I don't specify additional ones in the `IKOptions`)?

hongkai-dai, 2015-04-17:
Hmm, that is interesting, could you try instead of using time [0, 5], using
[0, 1], and fix the final posture at time 1, with three or 4 knot points?
We run inverseKinTraj a lot to compute the arm lifting motion, and we did
not observe that swing. Could you also post the code for calling
inverseKinTraj?

When you say "WorldOrientationConstraint", do you mean
"WorldQuatConstraint"?

steinachim, 2015-04-17:
Interesting... With t = 0 : 0.25 : 1 it seems to work just fine and even with many more knot points the swing is much less pronounced. So this might be a potential solution. But then I'd have to add time parameterization manually, which I had hoped to avoid. 

I added the relevant parts of the code to the gist (https://gist.github.com/steinachim/96af7881c64dde0a4c79) and yes, I meant `WorldQuatConstraint`.

On a related note: In MATLAB I receive a `PPTrajectory` which allows me to extract the positions and velocities at any time point I want. In the C++ implementation, the result is in the form of simple matrices for q, qd and qdd at the time-points t. Is there functionality provided somewhere to do the interpolation - especially in the extreme case of only 2 time-points?

hongkai-dai, 2015-04-17:
We just added the functionality of piecewise polynomial in c++, you can
check the pull request https://github.com/RobotLocomotion/drake/pull/972

hongkai-dai, 2015-04-17:
I think for the timing problem, the reason is that the cost function of
inverseKinTraj has three components, it contains the posture error, the
velocity penalty and the acceleration penalty, each component has its own
cost matrix, which is set in IKoptions.Q, Ikoptions.Qv, IKoptions.Qa.
These cost matrices are tuned for time [0,1]. When you scale the time,
these matrix should also be scaled accordinly, proportional to the time
scaling. For example when you use time [0,5], to reach the same final
posture, your joint velocity would be 5 times slower than that if you used
time [0,1], so the cost matrix IKoptions.Qv should be 5 times larger, and
IKoptions.Qa should be 25 times larger, if you want to achieve the same
behavior.

hongkai-dai, 2015-04-17:
Does changing cost matrices fix your problem? If it does, could you close this issue?

steinachim, 2015-04-18:
I just checked and it does indeed, thank you!

closed 2015-04-18 by steinachim