RobotLocomotion/drake #1006: MAKE TEST: LC singleton fail: java.net.SocketException: Can't assign requested address

tarquasso, 2015-04-18:
Running the following tests within Matlab were unsuccessful:
- StateMachineControl.run() in the examples/PlanarMonopodHopper directory ended with a warning.
- runLQR in the examples/Quadrotor directory caused the Matlab 2014b instance to crash.
 
Therefore I am currently running $make test in the drake-distro folder, it is already running for 2 hours. Many of the tests either fail or time out

It seems to have to at least partially have to do with one of these errors:

a) STARTUP ISSUE:
This indicates a potentially serious problem in your MATLAB setup, which should be resolved as soon as possible.  Error detected was:
MATLAB:run:FileNotFound
../../softrobotics-git/drake-distro/addpath_pods not found. 
> In matlabrc at 228 
To reproduce this test use rng(968673823,'twister’)

My matlab startup folder is:
/Users/rkk/Documents/Matlab
It contains a 'startup.m’ file with this content:
run('../../softrobotics-git/drake-distro/addpath_pods’)
the drake distro sits here:
/Users/rkk/softrobotics-git/drake-distro/
Is that startup.m file content wrong?

b) ALL NETWORK TRAFFIC BROADCAST TO NETWORK:
> In addpath_drake at 74
  In addpath_drake at 4 
 Added the lcm jar to your javaclasspath (found via cmake)
Warning: Currently all of your LCM traffic will be broadcast to the network, because:
  Route to multicast channel does not run through the loopback interface.
  Consider running: sudo /Users/rkk/softrobotics-git/drake-distro/drake/util/setup_loopback_multicast.sh

Shall I just run:  sudo /Users/rkk/softrobotics-git/drake-distro/drake/util/setup_loopback_multicast.sh ?

c) CAN’T ASSIGN ADDRESS:
> In checkDependency at 62
  In checkDependency at 73
  In collisionDetectTest at 19
  In collisionDetectSphereTest at 2
  In fevalPackageSafe at 21 
 Added the lcmgl jar to your javaclasspath (found via cmake)
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
LC singleton fail: java.net.SocketException: Can't assign requested address

Is that because I am on a hotel wifi, that does not allow LCM to broadcast? Or something else?

RussTedrake, 2015-04-18:
#2 - yes.  Follow the instructions displayed on the screen.

#3 - as i said, search for yosemite on the drake github issues, you should find an easy fix.

#1 is probably #3.

tarquasso, 2015-04-18:
Thanks a lot, so this fixed it:
a) STARTUP ISSUE by changing 'startup.m' content to:
warning('off','Drake:DisablingSimulinkAutosave')
run('<fullpathto>/drake-distro/addpath_pods')

b) ALL NETWORK TRAFFIC BROADCAST TO NETWORK: $sudo <pathto>/drake-distro/drake/util/setup_loopback_multicast.sh

-c) CAN’T ASSIGN ADDRESS by following #558

closed 2015-04-18 by tarquasso