RobotLocomotion/drake #1009: matlabpool / parpool Dependency Check

KeyDamo, 2015-04-19:
This is line 494 of sampledFiniteTimeVerification.m:
```if (matlabpool('size')==0) matlabpool; end```
The problem is that, reportedly, ```matlabpool``` [was replaced](http://stackoverflow.com/questions/19131126/how-do-i-create-a-parallel-loop) by ```parpool``` in R2013b of the Parallel Computing Toolbox.  I stumbled into this because I am using the R2015a.  The quick fix would be:
```matlab
try
  if (matlabpool('size')==0) matlabpool; end
catch
  if isempty(gcp('nocreate')) parpool; end
end
```
However, I know that ```matlabpool``` is called in other parts of Drake.  Perhaps the solution you prefer is something like expanding and using the ```checkDependency()``` function to select the command to use according the the user's version, since this is more like the customary way of doing things I'm used to see in Drake.  Let me know and if the fix you have in mind is "communicable" I'll be happy to do it myself.

RussTedrake, 2015-04-19:
yes - it would be great to update this.  According to http://www.mathworks.com/help/distcomp/release-notes.html#btw1_9f-1 it looks like we don't actually need to call parpool; it will get started automatically by `parfor` or `spmd`.

so it looks like a sufficient change for most (all?) of our use cases would be 

```
if (exist('matlabpool','file') && matlabpool('size')==0) matlabpool; end
```

that seems minimal enough that we probably don't need to add a helper routine for it.  based on:

```
russt-laptop% grep -R matlabpool .
./examples/Pendulum/test/testLCMControl.m:% this one is not playing nicely with matlabpool (at least not consistently).
./examples/Pendulum/test/testLCMControl.m:if (matlabpool('size')>0), matlabpool('close'); end
./examples/Pendulum/test/testLCMControl.m:%job = batch('javaaddpath(''drake.jar'');runPendLCMControl','Workspace',struct(),'matlabpool',0,...
./examples/Pendulum/test/testLCMControl.m:job = batch('runPendLCMControl','Workspace',struct(),'matlabpool',0);
./systems/@PolynomialSystem/sampledFiniteTimeReachability.m:% if (matlabpool('size')==0) matlabpool('open',4); end
./systems/@PolynomialSystem/sampledFiniteTimeVerification.m:  if (matlabpool('size')==0) matlabpool; end
./systems/@PolynomialSystem/sampledTransverseVerification.m:  if (matlabpool('size')==0) matlabpool; end
./systems/test/testLCMVisualizer.m:if (matlabpool('size')>0), matlabpool('close'); end
```

it looks like there is only a few places where this would have to change.  If you're willing to do it and submit a pull request, that's fantastic.  

Thanks!
- Russ

RussTedrake, 2015-04-19:
looking again, i'm embarrassed to say that the current sampledFiniteTimeVerification doesn't even call parfor (it was probably taken back out for compatibility).  i will clean it up quickly and submit the pull request.

KeyDamo, 2015-04-19:
OK I saw the commit. That was quick! (and also not communicable!) Thanks!
But I think that I better leave the closing of the issue to the maintainer (you?)

This is just to leave a complete record here - in case you or someone wants to remember this thing about the setting: 
*A parfor-loop runs on the existing parallel pool. If no pool exists, parfor will start a new parallel pool,* **unless the automatic starting of pools is disabled** *in your parallel preferences. If there is no parallel pool and parfor cannot start one, the loop runs serially in the client session.* Source: http://www.mathworks.com/help/distcomp/parfor.html

RussTedrake, 2015-04-19:
Yes - i read that, too.  But at some point we have to trust that if the user is advanced enough to be changing their default settings, then they'll be able to figure out that they need to start a matlab pool, too.

The issue will be closed after my commit gets merged.  Thx

PR #1010 (2015-04-21): update to support newer parallel computing toolbox syntax
resolves #1009
closed 2015-04-21 by RussTedrake