RobotLocomotion/drake #1013: solveLCPmex crashes on simple example

RussTedrake, 2015-04-20:
changed the top of `systems/plants/test/fallingBrickLCP.m` to 
```
function fallingBrickLCP

options.floating = 'quat';
options.use_new_kinsol = true;
options.terrain = RigidBodyFlatTerrain();
p = TimeSteppingRigidBodyManipulator('FallingBrickBetterCollisionGeometry.urdf',.01,options);
x0 = p.resolveConstraints([getRandomConfiguration(getManipulator(p));zeros(getNumVelocities(p),1)]);
```
and crash, bang, boom! in solveLCPmex.

psiorx, 2015-04-21:
This is a bigger issue than just solveLCPmex.  It involves TimeSteppingRigidBodyManipulator and solveLCP in general.  

The main problem is that the forward Euler step used to project the state and constraints forward in time is no longer valid with a quaternion floating base.

I'm working with Twan to adapt some code he had previously to perform a more appropriate integration on the state and constraints that keeps the quaternion floating base of unit length without introducing normalization error.

mposa, 2015-04-21:
I haven't really though at all about how this plays into the LCP solve, but I think the standard projection would be to follow the geodesic. I think, if you start at some quaternion q_0, and wish to follow the geodesic in direction u (with unit norm) for a distance of d, then you get the following update:

q_1 = q_0 + u*tan(d) 

and then normalize that vector

RussTedrake, 2015-09-27:
resolved by #1375

closed 2015-09-27 by RussTedrake