RobotLocomotion/drake #1046: runAtlasBalancing almost always ends badly.

RussTedrake, 2015-04-24:
the test doesn't fail with an error, but the toes point down while in the error and then things explode.

i think it's defunct.  most likely resolution: zap the test and rename runAtlasBalancingSplit to runAtlasBalancing (and same for the other tests with Split in the name).

rdeits, 2015-04-24:
I can't reproduce this. That's a problem that occurs when the support logic is not set correctly in the plan, but it's correct in the test file, and the execution looks good on my machine on RobotLocomotion/master. Are you sure you're up to date?

Regarding the naming, the test called `runAtlasBalancingSplit` is actually testing the multiple_contacts option. I've renamed it in #1048

psiorx, 2015-04-24:
Works fine for me as well.

RussTedrake, 2015-04-24:
definitely on master.  definitely fails every time for me.

rdeits, 2015-04-24:
What's your drake commit SHA?

RussTedrake, 2015-04-24:
i'll be in lab soon (can try to reproduce on my work laptop)

russt-laptop% git status
On branch master
Your branch is up-to-date with 'origin/master'.

nothing to commit, working directory clean
russt-laptop% git log -n 4
commit 2a937a295b4799f1d513d6f78e7fb3b4e091f2d9
Merge: 3dc299e 4755198
Author: Russ Tedrake <russt@mit.edu>
Date:   Fri Apr 24 14:34:48 2015 -0400

    Merge pull request #1041 from psiorx/jc-rigidbody-multicontact-fix

    Fix multi-contact on rigidbody distro

commit 3dc299e6fbf52bf351f76e83fe06aac0dca8df68
Merge: 5bc13e8 e6f58dd
Author: Russ Tedrake <russt@mit.edu>
Date:   Fri Apr 24 14:33:21 2015 -0400

    Merge pull request #1043 from RussTedrake/quick_fix

    remove unused lines in QPCommon

commit 5bc13e8b000420a770bca83062fa118daf5a690a
Merge: c9260e9 80c474b
Author: Russ Tedrake <russt@mit.edu>
Date:   Fri Apr 24 14:32:40 2015 -0400

    Merge pull request #1048 from rdeits/rd-rename-balancing-tests

    rename balancing tests for clarity

rdeits, 2015-04-24:
OK. I'm on the exact same commit and can't reproduce. Let's check it out in lab,.

commit b7ba151f (2015-04-26): fixed contactPhi.  resolves #1046 .
(might also help with #538)
closed 2015-04-26 by RussTedrake