RobotLocomotion/drake #1053: suspicious lines in RigidBodyManipulatorContact.cpp

RussTedrake, 2015-04-26:
was poking through RigidBodyManipulatorContact.cpp (quieting a few windows build warnings) and was surprised to see the lines:

139:     forwardJac(bodyInd - 1, bodyPoints, 0, J_tmp);

and 

163:   forwarddJac(bodyInd - 1, bodyPoints, dJ_tmp); //dJac instead of Jac

those look very suspicious to me.  I think it should be just bodyInd (not bodyInd-1, which only happens at mex interface levels when we convert matlab indices into c++ indices).   @psiorx - can you take a look at confirm for me?

psiorx, 2015-04-29:
The unit test would've caught something like this.

Since these indices are being pulled from idxA and idxB that are computed in the Matlab collisionDetect(), they are 1 based and need to be adjusted before being passed to the C++ methods which expect 0 based indices.

The code is correct but perhaps I can do the index adjustment higher upstream if convention calls for it.

RussTedrake, 2015-04-30:
any off-by-one conversions between matlab and c should definitely only happen in the mex interface files.  we want to be able to call these methods from the c++ parser, too.  thanks.

PR #1072 (2015-05-16): moved body index correction out of RBM and into the mex file
Resolves #1053
closed 2015-05-16 by RussTedrake