RobotLocomotion/drake #1062: Drake lcmtype c++ headers have wrong includes

rdeits, 2015-04-27:
Drake's c++ headers get installed to `build/include/lcmtypes/drake/`, so you would typically want to include on with:

```
#include "lcmtypes/drake/lcmt_qp_controller_input.hpp"
```

but the headers themselves reference other lcmtypes without the `lcmtypes` prefix. For example, the beginning of `lcmt_qp_controller_input.hpp` looks like:

```
#include "drake/lcmt_zmp_data.hpp"
#include "drake/lcmt_support_data.hpp"
#include "drake/lcmt_body_motion_data.hpp"
#include "drake/lcmt_body_wrench_data.hpp"
#include "drake/lcmt_whole_body_data.hpp"
#include "drake/lcmt_joint_pd_override.hpp"
```

This makes it impossible to include any Drake c++ headers without playing around with the includes path in cmake.

rdeits, 2015-05-01:
The workaround is to add the following to your CMakeLists:

```
include_directories (${CMAKE_BINARY_DIR}/lcmgen/lcmtypes )
```

RussTedrake, 2015-05-01:
looking at this quickly, I actually think that adding this include directory is the right solution (not just a temporary work-around).  but i can add it in lcmtypes.cmake , so it works no and forever.

rdeits, 2015-05-03:
I don't think I agree. Should the #include directives in the LCM types refer to the actual paths from the `includes/` folder? Also, the `include_directories` fix that I suggested doesn't work outside of the drake pod. Other pods have to instead set `include_directories (${CMAKE_INSTALL_PREFIX}/include/lcmtypes )` (but we *can't* set that in Drake, or a newly built c++ object will try to include headers from a previously installed (and possibly out of date) lcmtype)

rdeits, 2015-05-03:
(edited an important "should" to "shouldn't" in my prior comment)

RussTedrake, 2015-05-04:
For that case i think it's still ok;  but you're right that i need to ensure that the pod-build include takes priority over the build includes.

All of these problems stem from my attempt to resolve the perceived issue of lcmtypes effectively doing an in-source build (the lcm-gen generates the c files in place).  If i were to restore this functionality, then i think everything would again work as you expected.  Alternatively, i could more completely reproduce the directory hierarchy in the pod-build directory so it's all correct within a pod.

closed 2015-08-21 by RussTedrake