RobotLocomotion/drake #1064: matlabpool / parpool Dependency Check ...... 2

KeyDamo, 2015-04-28:
This is a continuation of issue #1009 / commit #1010.
The lines added in commit #1010 of the form
```matlab
if (exist('matlabpool','file') && matlabpool('size')==0) matlabpool; end
```
are throwing the error 
```
Error using matlabpool (line XXX)
matlabpool has been removed.
To query the size of an already started parallel pool, query the 'NumWorkers'
property of the pool.
To check if a pool is already started use 'isempty(gcp('nocreate'))'.
```
I checked that the ```matlabpool``` file still exists in the most recent versions of the Parallel Computing Toolbox.  In place of that line, I suggest using:
```matlab
distcompversion=ver('distcomp');
if ~isempty(distcompversion)
  if (str2num(v.Version)<6.3) matlabpool; end; end;
clear distcompversion
```

RussTedrake, 2015-05-02:
of course you're right that this file must exist (to print the deprecated message).  i should have thought of that.  here is my quick attempt to fix it, without requiring too much code in every file instance.

KeyDamo, 2015-05-03:
yes, I see that's a better solution (I didn't do enough from my side to be able to learn about ```verLessThan```)

commit 6d720ff0 (2015-05-04): Merge pull request #1080 from RussTedrake/distcomp_versions

resolves #1064
commit 6e915e04 (2015-05-04): resolves #1064
closed 2015-05-04 by RussTedrake