RobotLocomotion/drake #1073: getPelvisBodyMotionForWalking overwrites options that are passed in

manuelli, 2015-04-30:
```MATLAB
function pelvis_motion_data = getPelvisMotionForWalking(obj, foot_motion_data, supports, support_times, options)
% Get the BodyMotionData for the robot's pelvis, given the BodyMotionData for its feet, plus its planned support sequence. 
typecheck(foot_motion_data, 'BodyMotionData');
typecheck(supports, 'RigidBodySupportState');
assert(isnumeric(support_times));

if nargin < 6
  options = struct();
end
```

should be nargin < 5, am working on a branch that includes a fix for this.

RussTedrake, 2016-01-23:
it's a trivial fix.  can you please push?

manuelli, 2016-01-23:
I just looked at master and it looks like it is correct there. getPelvisMotionForWalking now also takes x0 as an argument, so 'options' is actually the 6th arg. 

https://github.com/RobotLocomotion/drake/blob/master/drake/systems/robotInterfaces/%40Biped/getPelvisMotionForWalking.m#L1

closed 2016-01-23 by RussTedrake