RobotLocomotion/drake #1096: a few atlas example failures on new buildserver

RussTedrake, 2015-05-08:
these happen reliably on the nightly tests

Site: thrashcan
Build Name: master-RobotLocomotion-master
Build Server Directory: /tmp/drake-master/drake-32276
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=wrong
site

Most recent commits to drake included in this build:
```
0c13a16 Merge pull request #1089 from hongkai-dai/test-IK-cpp
7da738c no snopt_cpp in CMakeLists anymore, it is called snopt_c
b667715 Merge branch 'master' of
https://github.com/RobotLocomotion/drake-private; branch 'master' of
https://github.com/RobotLocomotion/drake
6d720ff Merge pull request #1080 from RussTedrake/distcomp_versions
665ef45 Merge pull request #1065 from
hongkai-dai/add-bodySpatialMotionPDmex
```

403 tests run:
379 passed
0 cancelled (with missing dependencies )
12 timed out
12 failed
 2 <a href="https://github.com/RobotLocomotion/drake/issues/310">known
issue #310</a>: testLCMPlant fails on parallel build because it can't keep
up with realtime
 4 <a href="https://github.com/RobotLocomotion/drake/issues/374">known
issue #374</a>: failing imu unit tests
 1 <a href="https://github.com/RobotLocomotion/drake/issues/466">known
issue #466</a>: implement Jdot*v for drakeFunction.kinematic methods
 1 <a href="https://github.com/RobotLocomotion/drake/issues/585">known
issue #585</a>: testIndividualCentersOfPressure fails
 1 <a href="https://github.com/RobotLocomotion/drake/issues/723">known
issue #723</a>: runDRCPracticeTerrain errors out with  probability
 3 unknown

examples/Atlas/runAtlasBalancingWithContactSensor (unknown issue)
```
error_id = Simulink:SFunctions:SFcnErrorStatus
error_message = Error reported by S-function 'DCSFunction' in
'MIMOCascade_1936428/system1/system2/system2/system2/system2/DrakeSys':

DrakeSystem S-Function: error MATLAB:dimagree in MATLAB callback.
See additional debugging information above
is_segfault = 0
```
```Matlab

                           < M A T L A B (R) >
                 Copyright 1984-2014 The MathWorks, Inc.
                   R2014a (8.3.0.532) 64-bit (glnxa64)
                            February 11, 2014

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(1127269042,'twister')

Calling addpath_gurobi
Calling addpath_snopt
Calling addpath_spotless
[Warning: Map ptr is NULL.  Assuming flat terrain at z=0] 
[Warning: Map ptr is NULL.  Assuming flat terrain at z=0] 
In BulletModel::findClosestPointsBtwElements: No closest point found!
In BulletModel::findClosestPointsBtwElements: No closest point found!
In BulletModel::findClosestPointsBtwElements: No closest point found!
In BulletModel::findClosestPointsBtwElements: No closest point found!
In BulletModel::findClosestPointsBtwElements: No closest point found!
[... 538 lines cut ...]
Error using getada1
Out of memory. Type HELP MEMORY for your options.

Error in sedumi (line 452)
       ADA = getada1(ADA, A, Ablkjc(:,3), Aord.lqperm, d, K.qblkstart);

Error in msdp/msol (line 83)
[dual_vars,y,info] = sedumi(P.A,P.b,P.c,P.K,pars); % modified by Milan
Korda

Error in PolynomialProgram/gloptipoly (line 53)
 [exitflag,objval] = msol(prog);

Error in polyIK (line 38)
[x,objval,exitflag] = gloptipoly(prog);

Error in fevalPackageSafe (line 21)
 feval(f);

<test_name>examples/PlanarNLink/test/polyIK</test_name>
<error_id>MATLAB:nomem</error_id> <error_message>Out of memory. Type HELP
MEMORY for your options.</error_message> 
matlab exit code: 1

```

gizatt, 2015-05-08:
> 2 known
> 4 known
> issue #374 <https://github.com/RobotLocomotion/drake/issues/374>: failing
> 1 known

A common point is that those two tests use the old controller, which may
have stopped being properly supported and testing now that we've properly
switched over. Switching the contact sensor test over to the next split
controller fixed the issue; I'll switch the other over as well. The only
remaining tests that appear to use the old controller are runAtlasJumping
and runAtlasRunning... perhaps they're happier because the robot model
isn't being changed in those cases?

RussTedrake, 2015-05-09:
Thanks!  Wish we had a better understanding, but updating away from the deprecated controller definitely seems like the right thing to do.

RussTedrake, 2015-08-27:
i believe this is resolved (or at least covered by another issue).  haven't noticed it on the build servers lately.

closed 2015-08-27 by RussTedrake