RobotLocomotion/drake #1100: Error in runLQRTree on Pendulum

wclin892, 2015-05-10:
Dear Russ Tedrake:

Recently, I try to runLQRTree on Pendulum case development
folder  and I want to make it work , but I am have big problem in it.
In this case, there is an error message:

In an assignment  A(I) = B, the number of elements in B and I must be the same.

Error in NonlinearProgram/objectiveAndNonlinearConstraints (line 581)
          [f(f_count+(1:obj.nlcon{i}.num_cnstr)),G(f_count+(1:obj.nlcon{i}.num_cnstr),obj.nlcon_xind_stacked{i})] = ...

Error in NonlinearProgram/snopt/snopt_userfun (line 1314)
        [f,G] = objectiveAndNonlinearConstraints(obj,x_all);

Error in snoptUserfun (line 6)
  [f,G] = SNOPT_USERFUN(x);

Error in NonlinearProgram/snopt (line 1377)
        [x_free,objval,exitflag,xmul,Fmul] = NonlinearProgramSnoptmex(x0_free, ...

Error in NonlinearProgram/solve (line 973)
            [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);

Error in DirectTrajectoryOptimization/solveTraj (line 188)
      [z,F,info,infeasible_constraint_name] = obj.solve(z0);

Error in LQRTree.buildLQRTree (line 400)
        [xtraj,utraj,~,~,info] = prog.solveTraj(tf0);

Error in PendulumPlant/balanceLQRTree (line 151)
      c = LQRTree.buildLQRTree(obj,obj.xG,obj.uG,@()rand(2,1).*[2*pi;10]-[pi;5],Q,R,options);

Error in runLQRTree (line 6)
c = balanceLQRTree(pd);

Could I ask for you help to give me a guidance to resolve these problems?

Sincerely Yours

David

lavrinovu4, 2016-05-24:
Hi Russ, It seems the problem is with an initialization of constraint in LQRTree.m:
```matlab
prog = prog.addStateConstraint(FunctionHandleConstraint(0, 0,nX,@c.onTreeConstraint),nbreaks);
```

The size of low and upper bound is 1 in this case, and we have constraint only for one state, while Pendulum has two.

So the modified line fix this error in my case:
```matlab
prog = prog.addStateConstraint(FunctionHandleConstraint([0 0], [0 0],nX,@c.onTreeConstraint),nbreaks);
```

RussTedrake, 2016-11-27:
The LQRTree code in matlab needs updating, but will be replaced with a mature c++ implementation soon.

closed 2016-12-23 by RussTedrake