RobotLocomotion/drake #1103: bertini mpicc: No such file or directory when compiling drake-distro on master

tarquasso, 2015-05-12:
when running make in drake-distro, get the following compile error

-------------------------------------------
-- bertini
-------------------------------------------
make[2]: mpicc: No such file or directory
make[2]: *** [misc.o] Error 1
make[1]: *** [/Users/rkk/Documents/MATLAB/drake-distro/build/bin/bertini] Error 2
make: *** [all] Error 2

Found this in the readme of bertini:
"To compile Bertini in parallel, you will additionally need:
MPI implementation (e.g., MPICH, http://http://www.mpich.org/ or OpenMPI, http://www.open-mpi.org/)"
Shall I manually install OpenMPI or another MPI solution?

RussTedrake, 2015-05-12:
Rerun the install prereqs script (it will install mpich)

tarquasso, 2015-05-12:
thanks! that solved the problem!

Just to show what prereques were missing:
30-9-152:drake-distro rkk$ brew update
Already up-to-date.
30-9-152:drake-distro rkk$ ./install_prereqs.sh homebrew
Warning: pkg-config-0.28 already installed
Error: cmake-3.0.0 already installed
To install this version, first `brew unlink cmake'
Error: gtk+-2.24.24 already installed
To install this version, first `brew unlink gtk+'
installing prereqs for lcm
Error: wget-1.15 already installed
To install this version, first `brew unlink wget'
Error: xz-5.0.5 already installed
To install this version, first `brew unlink xz'
Error: glib-2.40.0_1 already installed
To install this version, first `brew unlink glib'
Error: coreutils-8.22 already installed
To install this version, first `brew unlink coreutils'
installing prereqs for libbot
Warning: jpeg-8d already installed
Error: cmake-3.0.0 already installed
To install this version, first `brew unlink cmake'
Error: gtk+-2.24.24 already installed
To install this version, first `brew unlink gtk+'
Error: libpng-1.6.12 already installed
To install this version, first `brew unlink libpng'
installing prereqs for eigen
installing prereqs for bullet
Warning: gnu-sed-4.2.2 already installed
installing prereqs for octomap
Error: cmake-3.0.0 already installed
To install this version, first `brew unlink cmake'
==> Downloading https://homebrew.bintray.com/bottles/doxygen-1.8.9.1.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring doxygen-1.8.9.1.yosemite.bottle.tar.gz
🍺  /usr/local/Cellar/doxygen/1.8.9.1: 5 files, 15M
installing prereqs for snopt
Warning: gcc-4.9.2_1 already installed
installing prereqs for gurobi
==> Downloading https://homebrew.bintray.com/bottles/curl-7.42.1.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring curl-7.42.1.yosemite.bottle.tar.gz
==> Caveats
This formula is keg-only, which means it was not symlinked into /usr/local.

Mac OS X already provides this software and installing another version in
parallel can cause all kinds of trouble.

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/curl/lib
    CPPFLAGS: -I/usr/local/opt/curl/include

==> Summary
🍺  /usr/local/Cellar/curl/7.42.1: 305 files, 2.9M
installing prereqs for bertini
==> Installing flex dependency: gettext
==> Downloading https://homebrew.bintray.com/bottles/gettext-0.19.4.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring gettext-0.19.4.yosemite.bottle.tar.gz
==> Caveats
This formula is keg-only, which means it was not symlinked into /usr/local.

Mac OS X provides similar software, and installing this software in
parallel can cause all kinds of trouble.

OS X provides the BSD gettext library and some software gets confused if both are in the library path.

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/gettext/lib
    CPPFLAGS: -I/usr/local/opt/gettext/include

==> Summary
🍺  /usr/local/Cellar/gettext/0.19.4: 1920 files, 21M
==> Installing flex
==> Downloading https://homebrew.bintray.com/bottles/flex-2.5.39.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring flex-2.5.39.yosemite.bottle.tar.gz
==> Caveats
This formula is keg-only, which means it was not symlinked into /usr/local.

Mac OS X already provides this software and installing another version in
parallel can cause all kinds of trouble.

Some formulae require a newer version of flex.

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/flex/lib
    CPPFLAGS: -I/usr/local/opt/flex/include

==> Summary
🍺  /usr/local/Cellar/flex/2.5.39: 45 files, 2.0M
==> Downloading https://homebrew.bintray.com/bottles/bison-3.0.4.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring bison-3.0.4.yosemite.bottle.tar.gz
==> Caveats
This formula is keg-only, which means it was not symlinked into /usr/local.

Mac OS X already provides this software and installing another version in
parallel can cause all kinds of trouble.

Some formulae require a newer version of bison.

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/bison/lib

==> Summary
🍺  /usr/local/Cellar/bison/3.0.4: 51 files, 2.2M
==> Downloading https://homebrew.bintray.com/bottles/mpich2-3.1.4.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring mpich2-3.1.4.yosemite.bottle.tar.gz
🍺  /usr/local/Cellar/mpich2/3.1.4: 850 files, 16M
installing prereqs for avl
Warning: gcc-4.9.2_1 already installed
installing prereqs for xfoil
Warning: gcc-4.9.2_1 already installed
installing prereqs for drake
java version >= 1.6: YES
Error: boost-1.55.0 already installed
To install this version, first `brew unlink boost'
Error: graphviz-2.36.0 already installed
To install this version, first `brew unlink graphviz'

closed 2015-05-12 by tarquasso