RobotLocomotion/drake #1106: Everything called twice per sim time step when running a FeedbackSystem

tkoolen, 2015-05-14:
This is due to the fact that both FeedbackSystem.update and FeedbackSystem.output call getOutputs on the FeedbackSystem, which calls output on each of the systems that make up the feedback loop. Simulink calls sequentially calls update and then output at each timestep.

To see this run e.g. runAtlasWalking and set a breakpoint in FeedbackSystem.getOutputs and inspect `t` and the stack traces when you hit the breakpoint.

Fixing this could result in up to a 2x speed increase when simulating using simulink (although kinematics caching mitigates the problem, probably significantly).

RussTedrake, 2015-05-16:
it's true that output is getting called twice.  i think this is necessary (though we could implement caching at the feedback system level).  i've looked more closely and there is one case where i could save a call to one of the outputs -- i've done that now.  

i think that the real problem you were seeing is that -- for the case of contact sensors (e.g. atlas) -- solveLCP was probably being called twice!  once with mex and once without.  and the caching was being violated.  i believe i've resolved that now, but don't actually have a good test case for this in mainline drake.  @gizatt -- could you implement the foot_force_sensors option in the mainline drake atlas when you have time?

gizatt, 2015-05-16:
Will do!

gizatt, 2015-06-26:
This is already implemented in examples/Atlas/runAtlasBalancingWithContactSensor, I think?

RussTedrake, 2015-06-26:
you're right.  i forgot.

closed 2015-06-26 by RussTedrake