RobotLocomotion/drake #1113: confused about installation

kagimbi, 2015-05-30:
can someone help me with the step of what to download? i download drake zip file but it is not working My matlab is 2013b and my computer is 32bit. 
Any hint would help
 thank you

sheim, 2015-06-04:
Hi,

There is a good guide to the installation [here](https://github.com/RobotLocomotion/drake/wiki/Installation-and-Quickstart).

Judging from the error message RussTedrake/drake#18, my guess is that you did a `git clone` on Drake, but you should actually clone [drake-distro](https://github.com/RobotLocomotion/drake-distro) (which contains drake as well). Try following the directions on the wiki linked above.  
If you still have problems, feel free to ask again. Btw, if you do need more assistance, mention your OS as well (I'm guessing win32?). Cheers!

kagimbi, 2015-06-04:
Hi,thank you for the response. I am not able to install a compiler on my matlab. My matlab find many compiler but it cant find where they are .I attached a world document as what is telling  me. yes My computer is windows 32Thank you .

Date: Wed, 3 Jun 2015 23:57:08 -0700

sheim, 2015-06-04:
Hi,

Just so you know, you can't attach files to github issues (even if you just answer by e-mail and attach it, it won't show up here). Also, please answer directly here on github: not sure why github isn't stripping the old part of the e-mail, but it makes the conversation cluttered.

I don't have a box running windows handy, so I can't directly troubleshoot this, but if matlab finds a compiler, it should be relatively straightforward to select one. See [here](http://www.mathworks.com/help/matlab/matlab_external/choose-c-or-c-compilers.html).  
Also you might want to use the same setup they tested, see [here](https://github.com/RobotLocomotion/drake/wiki/Cygwin).

If nothing works, post the error matlab gives you when you try to select the compiler. You might also want to try searching that error in [stackoverflow](http://stackoverflow.com/).

Good luck

kagimbi, 2015-06-07:
Hello ,I am confused about the installing the prerequisites.I downloaded the CMake, the java jdk , the windows visual2010 and Cgwin.Now i m stuck on this step: -do cd drake-distro, then run ./install_prereqs.sh cygwin to install pod-specific dependencies using the cygwin installer(https://github.com/RobotLocomotion/drake/wiki/Cygwin#installing-prerequisites)Thank you for the Help

Date: Thu, 4 Jun 2015 15:13:35 -0700

sheim, 2015-06-07:
Cygwin is a _command line_ or _terminal_ emulator, similar to what you would have on a Unix system like Ubuntu or OS X (mac).

If you're unfamiliar with this, you should probably download the [binary version](https://github.com/RobotLocomotion/drake/wiki/FromBinary), which is pre-compiled. All you have to do is start Matlab, navigate to the `drake-distro/drake/` folder, enter `addpath_drake` and you should be good to go. Test this out first and see if it works well for you. If you're just trying things out or following the edx course, this is all you need too.

If you eventually want change to the source-version and stay up-to-date and contribute, you should get comfortable with both the _command-line_ as well as with _git_. For git, I found [this course](https://www.udacity.com/course/how-to-use-git-and-github--ud775) to be pretty awesome.

kagimbi, 2015-06-09:
Hello,I got everything figure out now, Thank you for your help and i am enjoying the online course suggestion now

Date: Sun, 7 Jun 2015 14:07:53 -0700

sheim, 2015-06-09:
Excellent : )
In that case, you can also close this issue. Enjoy!

closed 2015-06-09 by RussTedrake