RobotLocomotion/drake #1118: Unable to Install

AeroGauti, 2015-06-11:
Hi, i tried to follow the instructions and do accordingly, but i find very difficult with building the collection and testing installation. Any guidance would be a great help. As i am a beginner, finding it very difficult. :(

sheim, 2015-06-11:
The basic guide is [here](https://github.com/RobotLocomotion/drake/wiki/Installation-and-Quickstart).
If you're new, you should probably first try out the binary version, which is pre-compiled and the easiest way to start.

If you need more help, you should post the specific problems you're having, as well as your OS.
Also, see #1113 if any of your questions are already answered there. Cheers!

AeroGauti, 2015-06-11:
Hi, thanks for your reply. Installed cmake, and others too, compiled few examples, its running but lot of warnings and errors... Will be back with specific problems.. Cheers!!!!!!

sheim, 2015-06-11:
Some warnings, like 'zero-crossing detected', are completely normal.
Also, many of the examples require additional libraries like SNOPT
which are not open-source and not included by default. In these cases,
drake will give you an error.

Bu it seems that the installation is working fine now? In which case,
you can close the issue (and for new problems, open a new issue).

Sent from my iPhone

AeroGauti, 2015-06-12:
How do i close a issue?

closed 2015-06-12 by RussTedrake