RobotLocomotion/drake #1120: Error

AeroGauti, 2015-06-12:
"  addpath_drake
Error using addpath_drake (line 24)
You must run make first (and/or add your pod
build/matlab directory to the matlab path)"

How do i resolve this?

sheim, 2015-06-16:
It looks like you've downloaded [drake-distro](https://github.com/RobotLocomotion/drake-distro), but not compiled it yet. After downloading, you need to go to the path of drake-distro from the terminal and run `make`, see [here](https://github.com/RobotLocomotion/drake/wiki/FromSource#build-the-collection). Or use the [binary installation](https://github.com/RobotLocomotion/drake/wiki/FromBinary), and then you shouldn't need to do anything before `addpath_drake`.

From #1118, it seemed that you had already correctly installed everything, can you confirm this? Or what did you mean with "compiled some examples"? I ask, because before making drake-distro, I don't think any of the examples should work, so this is puzzling me.

AeroGauti, 2015-06-19:
Hi, thanks for your reply.
I mixed up the files... will redo it from starting, and come back to you with errors. :)

Thanks.

closed 2015-08-27 by RussTedrake