RobotLocomotion/drake #1131: Compilation of rigidbody fails

aespielberg, 2015-06-26:
I was working on compiling and installing drake from rigidbody.  I got pretty far, up to the make step, but I got the error posted below.  Have any of you ever gotten this error?  I'm on Ubuntu 12.04 64-bit / MATLAB 2015a (I don't think that's relevant here) / g++/gcc 4.7.3.  I don't know if any of these settings might be the problem?  This problem is at 91% compilation in the 'drake' package.

[ 91%] Building CXX object systems/plants/viewer/CMakeFiles/drake_urdf_renderer.dir/drake_urdf_renderer.cpp.o
In file included from /usr/include/boost/iterator/interoperable.hpp:13:0,
                 from /usr/include/boost/iterator/iterator_facade.hpp:11,
                 from /usr/include/boost/filesystem/v3/path.hpp:28,
                 from /usr/include/boost/filesystem.hpp:35,
                 from /home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:12:
/usr/include/boost/type_traits/is_convertible.hpp: In instantiation of ‘const bool boost::detail::is_convertible_basic_impl<boost::filesystem3::directory_iterator&, boost::filesystem3::directory_iterator>::value’:
/usr/include/boost/type_traits/is_convertible.hpp:295:5:   required from ‘const bool boost::detail::is_convertible_impl<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator>::value’
/usr/include/boost/type_traits/is_convertible.hpp:418:1:   required from ‘struct boost::is_convertible<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator>’
/usr/include/boost/mpl/aux_/nested_type_wknd.hpp:26:31:   required from ‘struct boost::mpl::aux::nested_type_wknd<boost::is_convertible<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator> >’
/usr/include/boost/mpl/aux_/preprocessed/gcc/or.hpp:48:8:   required from ‘struct boost::mpl::or_<boost::is_convertible<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator>, boost::is_convertible<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator>, mpl_::bool_<false>, mpl_::bool_<false>, mpl_::bool_<false> >’
/usr/include/boost/iterator/detail/enable_if.hpp:68:12:   required from ‘struct boost::iterators::enable_if<boost::mpl::or_<boost::is_convertible<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator>, boost::is_convertible<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator>, mpl_::bool_<false>, mpl_::bool_<false>, mpl_::bool_<false> >, bool>’
/usr/include/boost/iterator/iterator_facade.hpp:67:12:   required from ‘struct boost::detail::enable_if_interoperable<boost::filesystem3::directory_iterator, boost::filesystem3::directory_iterator, bool>’
/usr/include/boost/iterator/iterator_facade.hpp:837:3:   required by substitution of ‘template<class Derived1, class V1, class TC1, class Reference1, class Difference1, class Derived2, class V2, class TC2, class Reference2, class Difference2> typename boost::detail::enable_if_interoperable<Derived1, Derived2, typename boost::mpl::apply2<boost::detail::always_bool2, Derived1, Derived2>::type>::type boost::operator!=(const boost::iterator_facade<Derived1, V1, TC1, Reference1, Difference1>&, const boost::iterator_facade<Derived2, V2, TC2, Reference2, Difference2>&) [with Derived1 = boost::filesystem3::directory_iterator; V1 = boost::filesystem3::directory_entry; TC1 = boost::single_pass_traversal_tag; Reference1 = boost::filesystem3::directory_entry&; Difference1 = long int; Derived2 = boost::filesystem3::directory_iterator; V2 = boost::filesystem3::directory_entry; TC2 = boost::single_pass_traversal_tag; Reference2 = boost::filesystem3::directory_entry&; Difference2 = long int]’
/usr/include/boost/filesystem/v3/operations.hpp:691:49:   required from here
/usr/include/boost/type_traits/is_convertible.hpp:136:49: error: use of deleted function ‘boost::filesystem3::directory_iterator::directory_iterator(const boost::filesystem3::directory_iterator&)’
In file included from /usr/include/boost/filesystem.hpp:36:0,
                 from /home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:12:
/usr/include/boost/filesystem/v3/operations.hpp:581:9: note: ‘boost::filesystem3::directory_iterator::directory_iterator(const boost::filesystem3::directory_iterator&)’ is implicitly deleted because the default definition would be ill-formed:
/usr/include/boost/filesystem/v3/operations.hpp:581:9: error: use of deleted function ‘boost::shared_ptr<boost::filesystem3::detail::dir_itr_imp>::shared_ptr(const boost::shared_ptr<boost::filesystem3::detail::dir_itr_imp>&)’
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:10:
/usr/include/boost/smart_ptr/shared_ptr.hpp:168:25: note: ‘boost::shared_ptr<boost::filesystem3::detail::dir_itr_imp>::shared_ptr(const boost::shared_ptr<boost::filesystem3::detail::dir_itr_imp>&)’ is implicitly declared as deleted because ‘boost::shared_ptr<boost::filesystem3::detail::dir_itr_imp>’ declares a move constructor or move assignment operator
In file included from /usr/include/boost/iterator/interoperable.hpp:13:0,
                 from /usr/include/boost/iterator/iterator_facade.hpp:11,
                 from /usr/include/boost/filesystem/v3/path.hpp:28,
                 from /usr/include/boost/filesystem.hpp:35,
                 from /home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:12:
/usr/include/boost/type_traits/is_convertible.hpp:128:41: error:   initializing argument 1 of ‘static boost::type_traits::yes_type boost::detail::checker<T>::_m_check(T, int) [with T = boost::filesystem3::directory_iterator; boost::type_traits::yes_type = char]’
In file included from /usr/include/c++/4.7/bits/stl_algobase.h:65:0,
                 from /usr/include/c++/4.7/bits/char_traits.h:41,
                 from /usr/include/c++/4.7/ios:41,
                 from /usr/include/c++/4.7/ostream:40,
                 from /usr/include/c++/4.7/iostream:40,
                 from /home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:1:
/usr/include/c++/4.7/bits/stl_pair.h: In instantiation of ‘constexpr std::pair<_T1, _T2>::pair(_U1&&, const _T2&) [with _U1 = std::pair<int, char*>; <template-parameter-2-2> = void; _T1 = std::pair<int, char*>; _T2 = boost::shared_ptr<Link>]’:
/usr/include/c++/4.7/bits/stl_pair.h:273:72:   required from ‘constexpr std::pair<typename std::__decay_and_strip<_Tp>::__type, typename std::__decay_and_strip<_T2>::__type> std::make_pair(_T1&&, _T2&&) [with _T1 = std::pair<int, char*>; _T2 = boost::shared_ptr<Link>&; typename std::__decay_and_strip<_T2>::__type = boost::shared_ptr<Link>; typename std::__decay_and_strip<_Tp>::__type = std::pair<int, char*>]’
/home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:337:87:   required from here
/usr/include/c++/4.7/bits/stl_pair.h:126:45: error: use of deleted function ‘boost::shared_ptr<Link>::shared_ptr(const boost::shared_ptr<Link>&)’
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:10:
/usr/include/boost/smart_ptr/shared_ptr.hpp:168:25: note: ‘boost::shared_ptr<Link>::shared_ptr(const boost::shared_ptr<Link>&)’ is implicitly declared as deleted because ‘boost::shared_ptr<Link>’ declares a move constructor or move assignment operator
/usr/include/boost/smart_ptr/shared_ptr.hpp: In instantiation of ‘boost::shared_ptr<T>& boost::shared_ptr<T>::operator=(const boost::shared_ptr<T>&) [with T = Geometry; boost::shared_ptr<T> = boost::shared_ptr<Geometry>]’:
/home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:173:14:   required from here
/usr/include/boost/smart_ptr/shared_ptr.hpp:305:9: error: use of deleted function ‘boost::shared_ptr<Geometry>::shared_ptr(const boost::shared_ptr<Geometry>&)’
/usr/include/boost/smart_ptr/shared_ptr.hpp:168:25: note: ‘boost::shared_ptr<Geometry>::shared_ptr(const boost::shared_ptr<Geometry>&)’ is implicitly declared as deleted because ‘boost::shared_ptr<Geometry>’ declares a move constructor or move assignment operator
In file included from /usr/include/c++/4.7/list:64:0,
                 from /home/aespielberg/drake/drake-distro/drake/systems/plants/viewer/drake_urdf_renderer.cpp:3:
[... cut ...]
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.7/README.Bugs> for instructions.
Preprocessed source stored into /tmp/ccI0MtcK.out file, please attach this to your bugreport.
make[4]: *** [systems/plants/viewer/CMakeFiles/drake_urdf_renderer.dir/drake_urdf_renderer.cpp.o] Error 1
make[3]: *** [systems/plants/viewer/CMakeFiles/drake_urdf_renderer.dir/all] Error 2
make[2]: *** [all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

RussTedrake, 2015-06-26:
I haven't seen this one before.  Based on a quick check, it looks like maybe it's a boost issue?
http://stackoverflow.com/questions/18900730/boostshared-ptrshared-ptrconst-boostshared-ptr-is-implicitly-declared

we do have a build server still running 12.04, but it has an older version of matlab, too.  perhaps we haven't compiled your setup exactly.

that file is actually the last piece of drake that still has a boost dependency, and we aim to zap even that.  it looks like it would be pretty trivial to convert that file to std::shared_ptrs and spruce for the filenames.  @psiorx, might you take a crack at it?

- Russ

psiorx, 2015-06-26:
Will do.

aespielberg, 2015-06-26:
I see.  What is spruce (or did you mean spruce it up)?  In my version,
should I just try changing all of the boost:: shared pointers to std::?

-Andy S.

Andrew Spielberg
PhD Student
MIT - Computer Science and Artificial Intelligence Laboratory

RussTedrake, 2015-06-26:
there is a pull request (which you could merge locally) here:
https://github.com/RobotLocomotion/drake/pull/1133

it will be pulled into master probably later tonight -- after the build servers approve

aespielberg, 2015-06-27:
Okay great - I won't have much time tonight so if it will be up in the
master by tomorrow morning I'll just do that.  Thanks a ton!

RussTedrake, 2015-06-27:
It's in (drake) master.  Cd drake.  Git checkout master.  Git pull.  Make.

closed 2015-06-27 by RussTedrake

aespielberg, 2015-06-29:
Thanks a ton for doing this so quickly!  It built fine and I was able to
run and complete most of the tests!  I noticed at the end it said that it
was sending test results via http over to you guys...maybe that will be
useful if this configuration is new.  *shrugs*

-Andy S.

RussTedrake, 2015-06-29:
Glad it worked.

PR #1133 (2015-06-27): removed boost dependency from drake viewer
This should be the last of the boost dependencies. 

Boost smart pointers were replaced with the standard library equivalent and boost file system calls were replaced with spruce.

Also fixed a drake viewer unit test.

Resolves #1131