RobotLocomotion/drake #1136: Visualizer unable to handle parameters

aespielberg, 2015-07-04:
Hi all,

I am trying to parameterize my robot, and in order to see what I'm doing better, I'm trying to parameterize the visualization as well.  Unfortunately, when I try to run the visualization:

%p is a rigidbodymanipulator, xtraj is a trajectory:

v2 = p.constructVisualizer()
v2.playback(xtraj);

I get the following error:

msspoly value cannot be converted to type of field "position" in Java class
drake.lcmt_viewer_geometry_data

Error in RigidBodyCylinder/serializeToLCM (line 71)
      geometry.position = obj.T(1:3,4);

Error in BotVisualizer (line 81)
            link.geom(j) = serializeToLCM(b.visual_geometry{j});

Error in RigidBodyManipulator/constructVisualizer (line 1644)
            v = feval(type,arg{:});

Error in runSwingUpKinematicTest (line 9)
v2 = p.constructVisualizer()

Is there a clean way that I'm supposed to handle this?  Otherwise, what is the most graceful way for me to edit this?  I'm pasting my .urdf below, which is just a slightly modified Acrobot.

<?xml version="1.0"?>

<robot xmlns="http://drake.mit.edu"
 xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
 xsi:schemaLocation="http://drake.mit.edu ../../../doc/drakeURDF.xsd" name="Acrobot">

  <parameter name="b1" value=".1" lb="0" />
  <parameter name="b2" value=".1" lb="0" />
  <parameter name="lc1" value=".5" lb="0" />
  <parameter name="lc2" value="5" lb="0" />
  <parameter name="Ic1" value=".083" lb="0" />
  <parameter name="Ic2" value=".33" lb="0" />

  <link name="base_link">
    <visual>
      <geometry>
         <box size=".2 .2 .2" />
      </geometry>
      <material name="green">
	<color rgba="0 1 0 1" />
      </material>
    </visual>
  </link>

  <link name="upper_link">
    <inertial>
      <origin xyz="0 0 -$lc1" rpy="0 0 0" />
      <mass value="1" />
      <inertia ixx="1" ixy="0" ixz="0" iyy="$Ic1" iyz="0" izz="1" />
    </inertial>
    <visual>
      <origin xyz="0 0 -$lc1" rpy="0 0 0" />
      <geometry>
         <cylinder length="1.1" radius=".05" />
      </geometry>
      <material name="red">
	<color rgba="1 0 0 1" />
      </material>
    </visual>
  </link>

  <link name="lower_link">
    <inertial>
      <origin xyz="0 0 -$lc2" rpy="0 0 0" />
      <mass value="1" />
      <inertia ixx="1" ixy="0" ixz="0" iyy="$Ic2" iyz="0" izz="1" />
    </inertial>
    <visual>
      <origin xyz="0 0 -$lc2" rpy="0 0 0" />
      <geometry>
         <cylinder length="2.1" radius=".05" />
      </geometry>
      <material name="blue">
	<color rgba="0 0 1 1" />
      </material>
    </visual>
  </link>
  
  <joint name="shoulder" type="continuous">
    <parent link="base_link" />
    <child link="upper_link" />
    <origin xyz="0 .15 0" />
    <axis xyz="0 1 0" />
    <dynamics damping="$b1" />
  </joint>

  <joint name="elbow" type="continuous">
    <parent link="upper_link" />
    <child link="lower_link" />
    <origin xyz="0 .1 -1" />
    <axis xyz="0 1 0" />
    <dynamics damping="$b2" />
  </joint>

  <transmission type="SimpleTransmission" name="elbow_trans">
    <actuator name="elbow_torque" />
    <joint name="elbow" />
    <mechanicalReduction>1</mechanicalReduction>
  </transmission>
</robot>

RussTedrake, 2015-07-07:
this wasn't something I had thought about before.  there was no real reason that it couldn't be done.  the two changes required were:
- the visualizers cache some information about geometry.  i've added a specific method to updateManipulator which you'll have to call to effectively flush that cache.  i've tested it for the planar visualizer and the libbot visualizer and it works fine.
- the visual and collision geometries in RigidBody didn't support parameters, but they do now.

i've added a simple test to the end of paramExtractFromURDF which changes the length of the first link of the acrobot and you can watch it grow in the visualizer.

aespielberg, 2015-07-08:
Wow, thank you so much for adding this functionality so quickly!

aespielberg, 2015-07-08:
I guess I"ll close this for now since it seems to work.  I'll let you know if there are any more bugs.  Thanks again.

closed 2015-07-08 by aespielberg