RobotLocomotion/drake #1137: Unable to parameterize joint locations

aespielberg, 2015-07-04:
Is parameterizing joint locations something that is supported?  For instance, If I want to parse the following URDF:

<?xml version="1.0"?>

<robot xmlns="http://drake.mit.edu"
 xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
 xsi:schemaLocation="http://drake.mit.edu ../../../doc/drakeURDF.xsd" name="Acrobot">

  <parameter name="b1" value=".1" lb="0" />
  <parameter name="b2" value=".1" lb="0" />
  <parameter name="Ic1" value=".083" lb="0" />
  <parameter name="Ic2" value=".33" lb="0" />
  <parameter name="length1" value="1.0" lb="0" />
  <parameter name="length2" value="2.0" lb="0" />

  <link name="base_link">
    <visual>
      <geometry>
         <box size=".2 .2 .2" />
      </geometry>
      <material name="green">
	<color rgba="0 1 0 1" />
      </material>
    </visual>
  </link>

  <link name="upper_link">
    <inertial>
      <origin xyz="0 0 -$length1 / 2" rpy="0 0 0" />
      <mass value="1" />
      <inertia ixx="1" ixy="0" ixz="0" iyy="$Ic1" iyz="0" izz="1" />
    </inertial>
    <visual>
      <origin xyz="0 0 -.5" rpy="0 0 0" />
      <geometry>
         <cylinder length="1.1" radius=".05" />
      </geometry>
      <material name="red">
	<color rgba="1 0 0 1" />
      </material>
    </visual>
  </link>

  <link name="lower_link">
    <inertial>
      <origin xyz="0 0 -$length2 / 2" rpy="0 0 0" />
      <mass value="1" />
      <inertia ixx="1" ixy="0" ixz="0" iyy="$Ic2" iyz="0" izz="1" />
    </inertial>
    <visual>
      <origin xyz="0 0 -1" rpy="0 0 0" />
      <geometry>
         <cylinder length="2.1" radius=".05" />
      </geometry>
      <material name="blue">
	<color rgba="0 0 1 1" />
      </material>
    </visual>
  </link>
  
  <joint name="shoulder" type="continuous">
    <parent link="base_link" />
    <child link="upper_link" />
    <origin xyz="0 .15 0" />
    <axis xyz="0 1 0" />
    <dynamics damping="$b1" />
  </joint>

  <joint name="elbow" type="continuous">
    <parent link="upper_link" />
    <child link="lower_link" />
    <origin xyz="0 .1 -$length1" />
    <axis xyz="0 1 0" />
    <dynamics damping="$b2" />
  </joint>

  <transmission type="SimpleTransmission" name="elbow_trans">
    <actuator name="elbow_torque" />
    <joint name="elbow" />
    <mechanicalReduction>1</mechanicalReduction>
  </transmission>
</robot>

Here I have parameterized the origin of the elbow joint.  In essence, I want my robot to be a continuous kinematic chain, and if one of the robot link lengths is a parameter, then the joint location will also have to be a parameter.

When I try to parse this in, however, I get the following error:

ndefined function 'any' for input arguments of type 'msspoly'.

Error in RigidBodyManipulator/addJoint (line 451)
      if any(xyz)

Error in RigidBodyManipulator/addRobotFromURDF>parseJoint (line 369)
model =
addJoint(model,name,type,parent,child,xyz,rpy,axis,damping,coulomb_friction,static_friction,coulomb_window,limits);

Error in RigidBodyManipulator/addRobotFromURDF>parseRobot (line 104)
  model = parseJoint(model,robotnum,joints.item(i),options);

Error in RigidBodyManipulator/addRobotFromURDF (line 54)
model = parseRobot(model,robot,xyz,rpy,options);

Error in RigidBodyManipulator (line 72)
          obj = addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);

RussTedrake, 2015-07-07:
i've fixed this in the pull request referenced above and confirmed that it works.

closed 2015-07-08 by RussTedrake

aespielberg, 2015-07-08:
This seems to work, thanks a ton!!

PR #1143 (2015-07-08): adds a parameter which modifies the kinematics to the test case. 
fixes a bug in passing msspoly's through the old wrl parser.
adds an extra test to acrobot parameter test.
resolves #1137