RobotLocomotion/drake #1138: Difficulty understanding ForwardKin

aespielberg, 2015-07-05:
Hi all,

I have the following problem: I have my robot in a position/velocity state.  I am trying to compute the forward kinematics of the object.  In particular, my goal is, given a point in a frame on the robot, and given the robot at a certain state, compute that point's location in the world frame .

Working with the AcrobotWParams.urdf, I am trying to use ForwardKin to do this.  However, I'm not sure if this is what ForwardKin is meant to do. I'm having trouble understanding precisely what's meant in the documentation.  I am not getting the right answers out as I change the properties of the plant.  An example:

First, using the standard AcrobotWParams.urdf file:

```

frame_name = 'lower_link';
state = [pi/4; pi/3; 0; 0];
point_in_frame = [0; 0; 0]; %get the origin of the lower_link in the world frame

try
    frameId = plant.findLinkId(frame_name);
catch
    frameId = plant.findFrameId(frame_name);
end

state_length = length(state);
%first half of state is position, second half is velocity;
q = state(1:state_length/2);
qd = state((state_length/2 + 1):end);

kinsol = plant.doKinematics(q, qd); 

options.rotation_type = 1;

[x, J] = plant.forwardKin(kinsol, frameId, point_in_frame, options);
```

Output:
```
x =

   -0.7071
    0.2500
   -0.7071
    0.0000
    0.7854
    0.0000
```

But, let's say I set the parameter lc2 to be 100.0.  Then I get the same answer.

Am I misunderstanding where this point is measured?  If the origin of the lower_link is now set to [0; 0; -100], shouldn't [0; 0; 0] in the lower_link's frame return something on that magnitude?  Where is point_in_frame measured with respect to?

RussTedrake, 2015-07-07:
the code is right, but `lc2` doesn't change anything about the kinematics.  lc2 just effects the center of mass of the second link, and the kinematics are always in body coordinates (relative to the parent link, not relative to the center of mass).  

if you take my updated example, where `l1` is also a parameter, then you'll see the change as you expect.

closed 2015-07-07 by RussTedrake

aespielberg, 2015-07-08:
Aha, I understand now.  I thought origin was relative to the COM.  Thank you for the clarification.

Jendker, 2019-08-15:
If I may ask in the same old thread:
what would be the proper way to reproduce this code using the new interface?

I would like to calculate forward kinematics of plant using python, but forwardKin has been split in one of the changes.