RobotLocomotion/drake #1139: Constraints with optional parameters?

aespielberg, 2015-07-05:
Hi all,

A quick question - looking for confirmation if this is the correct way to do the following:

Cost functions expect to take the input form (h, x, u) (for running costs) or (T, x) (for final costs).  Similarly, FunctionHandleConstraints expect inputs of the form (x).

If we are implementing our own solver, and our variable matrix z now has additional variables (v), is the correct way to incorporate these additional variables into running costs, final costs, and constraints so they can take the input form:

(h, x, u, v), (T, x, v), (x, v), respectively

to simply override the addConstraint methode in the solver?  Or is there a way to do this without overriding the method?  Is there anywhere you guys could point to where this has been done before as an example?

RussTedrake, 2015-07-07:
i would not recommend doing it this way.  NonlinearProgram has a method `addDecisionVariable` which will allow you to append your extra parameters to the decision parameter vector `x`.  The part that you have to do is simply in your cost function and constraints to break apart the x vector into your pieces and evaluate your cost, etc, correctly.

aespielberg, 2015-07-09:
Hi Russ,

Yep, this is what I had already done, but I am more curious about the cost and constraint functions themselves.  In, for example, DirtranTrajectoryOptimization.m, z is already parsed into x(t) when adding costs.  Constriants can work on all time steps but only for the state variables x.  I am wondering if functionality already exists that doesn't do the parsing; i.e. lets me work directly on z, or if I should just override DirtranTrajectoryOptimizaton's addcost function.

aespielberg, 2015-07-09:
Oh, or did you mean addCost and addConstraint, instead of addRunningCost and addStateConstraint?

aespielberg, 2015-07-09:
Okay, yes, I have figured out how to add constraints using the arbitrary z vector, thank you!

closed 2015-07-09 by aespielberg