RobotLocomotion/drake #1140: Setting parameters with kinematics?

aespielberg, 2015-07-06:
Hi all,

Sorry for all the posts.  This is the last one for now; it might be related to the previous one.

I have the following plant:

```
<?xml version="1.0"?>

<robot xmlns="http://drake.mit.edu"
 xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
 xsi:schemaLocation="http://drake.mit.edu ../../../doc/drakeURDF.xsd" name="Acrobot">

  <parameter name="length" value="1.0" lb="0" />

  <link name="base_link">
    <visual>
      <geometry>
         <box size=".2 .2 .2" />
      </geometry>
      <material name="green">
	<color rgba="0 1 0 1" />
      </material>
    </visual>
  </link>

  <link name="upper_link">
    <inertial>
      <origin xyz="0 0 -$length / 2" rpy="0 0 0" />
      <mass value="1" />
      <inertia ixx="1" ixy="0" ixz="0" iyy="1" iyz="0" izz="1" />
    </inertial>
    <visual>
      <origin xyz="0 0 -0.5" rpy="0 0 0" />
      <geometry>
         <cylinder length="1.1" radius=".05" />
      </geometry>
      <material name="red">
	<color rgba="1 0 0 1" />
      </material>
    </visual>
  </link>

 
  
  <joint name="shoulder" type="continuous">
    <parent link="base_link" />
    <child link="upper_link" />
    <origin xyz="0 0 0" />
    <axis xyz="0 1 0" />
    <dynamics damping="1" />
  </joint>

  <transmission type="SimpleTransmission" name="shoulder_trans">
    <actuator name="shoulder_torque" />
    <joint name="shoulder" />
    <mechanicalReduction>1</mechanicalReduction>
  </transmission>
</robot>
```

It is a simple Pendulum with a parameterized arm length (modulo the visualization).

Now, I have this function:

```
function [f, df] = positionDiff( plant, params, frame_name, point_in_frame, target_point, state)
%plant: the plant
%state: the current state of the plant
%params: the plant's current parameters
%Create a constraint function from inputs:
%frame_name: the name of the frame in which our constrained point lies
%point_in_frame: the location of the point within that frame
%target_point: the target point where we want this to go
%outputs: the difference, f [Del_x, del_y, del_z, del_r, del_p, del_y] and its Jacobian as df

%findFrameId: it's one of these, so either the try or the catch should work

plant = plant.setParams(params);
plant.getParams()
plant = plant.compile();

try
    frameId = plant.findLinkId(frame_name);
catch
    frameId = plant.findFrameId(frame_name);
end

state_length = length(state);
%first half of state is position, second half is velocity;
q = state(1:state_length/2);
qd = state((state_length/2 + 1):end);

kinsol = plant.doKinematics(q, qd); %TODO: consider options in the future

options.rotation_type = 1;

[x, J] = plant.forwardKin(kinsol, frameId, point_in_frame, options); %TODO: what does final argument, dj actually do?

f = target_point - x;
df = -J; %df/dx = d(target_point)/dq - df/dq = 0 - J = -J

x
end

```

I call it like so:

```
positionDiff( pl, params, 'upper_link', [0; 0; 0.5], [1; 1; 1; 1; 1; 1], [pi/4; 0])
```

Where my plant is pl, the params is just the length=1.0.  When I do this, I get that the arm COM is at:

```

x =

    0.3536
    0.0000
    0.3536
    0.0000
    0.7854
    0.0000

```

But if I call it again with params with length = 0.5, I get the exact same answer.

I'm fairly certain this isn't supposed to happen.  Any ideas as to why this may be happening?

psiorx, 2015-07-06:
This looks like a bug.  Thanks for finding it.

The kinematics are currently stored in a tree of transforms.  There is a Ttree matrix associated with each link that keeps track of that link's contribution to the kinematics.  If the geometry of the manipulator is based on parameters, the URDF parser should initialize this matrix correctly.

However, when updating the parameters after the object has been constructed, the Ttree matrix is not properly recomputed.  This should happen in the updateParams() method of RigidBody.

If you change the value of the parameter in the URDF and recreate your manipulator from it, you should get the correct result.

aespielberg, 2015-07-06:
Hi John,

Thank you for the reply.  For my application, manually editing the urdf won't be an option.  I will look at the updateParams() method to see if I can fix it, but if you guys know how to quickly fix it and are planning on working on it, I can wait for that push.

psiorx, 2015-07-06:
I might not be able to get to this immediately but I can give some advice on where I would start.

In addRobotFromURDF.m, there's a parseJoint() method.  This method looks at the origin element from the urdf and parses the initial value of the parameter($length in your case) into the xyz position of the joint's origin.  It then calls addJoint on the underlying RigidBodyManipulator with the numerical value of xyz.  Inside addJoint(), the Ttree matrix is initialized as a standard homogeneous transformation:

```MATLAB
 child.Ttree = [rpy2rotmat(rpy), xyz; 0,0,0,1];

Where xyz has the value of the parameter in it.  Something like this would have to happen in updateParams.

aespielberg, 2015-07-06:
Okay, cool.  I will give this a try this evening.  If I get it working I
will let you guys know and I can commit it if it's useful, although I can't
promise I know Drake well enough to know how to solve these problems in the
most efficient ways.

-Andy S.

Andrew Spielberg
PhD Student
MIT - Computer Science and Artificial Intelligence Laboratory

aespielberg, 2015-07-07:
Hi John,

I just spent some time reading the code and trying to play around with this.  I want to ask a question, and then I want to see if something that is happening is the correct behavior.

First, joints are not explicitly stored in the body, are they?  At least, I can't find them.  It seems a body is held with its parent joint.  So, this "child" in the above example, is in fact a body, right?

Well, I load my urdf as above as variable plant:

```
>> plant.body(2).linkname

ans =

upper_link

>> plant.body(2).jointname

ans =

shoulder
```

Makes sense.  So far so good.

But:

```
plant.body(2).Ttree

ans =

     1     0     0     0
     0     1     0     0
     0     0     1     0
     0     0     0     1
```

No matter what I set as my length parameter <b>in the URDF</b>, I get this output - that's even before I try to set anything directly in the code.  If I'm understanding this correctly, Ttree is the homogeneous transform between origin of the prior link in the tree and the origin of the previous joint, and T_body_to_joint is the transform between the prior joint and the current body (even though the wording says otherwise).  Then, that would mean two transforms need to have their parameters updated, no?  Also, why are these transforms unaffected by my URDF changing?  I think I'm having trouble understanding how everything is calculated.

Additional test: I can't seem to get the behavior in forwardKin to change even when I change the URDF, as you suggest it should.

aespielberg, 2015-07-07:
Followup: I DID notice that plant.body(2).com is changing as I change the urdf.  So something IS changing...just not that forwardKin is producing.

RussTedrake, 2015-07-07:
this is the same problem #1138.  you're changing the center of mass only.  forwardKin is in body coordinates, relative to the joint (not the com).  changing the com of a body has no effect on the kinematics.  you would have wanted to put the length in the position of the joint, as I've done in 
AcrobotWParams.urdf in 
https://github.com/RobotLocomotion/drake/pull/1143/files

closed 2015-07-07 by RussTedrake