RobotLocomotion/drake #1149: '-std=c++11' compilation error with matlab 2015a on ubuntu

RussTedrake, 2015-07-09:
in the spotless pod on matlab 2015a on ubuntu 12.04, I just saw
```
cc1plus: error: unrecognized command line option '-std=c++11'
```

RussTedrake, 2015-07-09:
note that I am running gcc 4.6.3:
```
gcc (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

and it looks like matlab officially only supports >= 4.7 now:
http://www.mathworks.com/support/compilers/R2015a/index.html?sec=glnxa64
and they officially only support ubuntu 14.04.

my guess is that changing the flag in `~/.matlab/R2015a/mex_C++_glnxa64.xml` from `-std=c++11` to `-std=c++01` (in two places) would probably resolve it.  or better yet, upgrading to gcc 4.7 .

RussTedrake, 2015-07-09:
also note that spotless is optional (and drake compiled fine).  if you don't need it, you could comment it out in the `drake-distro/tobuild.txt` file.

MiaoLi, 2015-07-11:
I had the same problem, ubuntu 12.04+ MATLAB 2015a .
I have gcc 4.7 and 4.8 installed, But I finally complied them by the following step:
cp /usr/local/MATLAB/R2015a/bin/mexopts.sh ~/.matlab/R2015a/mexopts.sh
Change CC=’gcc’ to CC=’gcc-4.7′ CCX=’g++’ to CCX=’g++-4.7′
chmod 644 ~/.matlab/R2013b/mexopts.sh
-----
Then change -std=c++11 to  -std=c++0x in CMakeList.

RussTedrake, 2015-07-11:
great.  thank you for reporting your findings!

RussTedrake, 2015-08-27:
we have officially updated the minimal gcc version to 4.7, making this one irrelevant.

closed 2015-08-27 by RussTedrake

ashutosh-devel, 2016-12-13:
To solve this:

go to the glnxa64/ directory inside the MATLAB/R20XXa/bin folder and replace all occurrences of c++11 with c++0x in the g++_glnax64.xml file present there. It worked for me.