RobotLocomotion/drake #1151: Build Error - libbot gdk/gdkx.h missing

vikrantvm, 2015-07-10:
I am trying to build from source, and following instructions on the wiki.I am using homebrew. I get the following error while running make:

```
drake-distro/libbot/bot2-vis/src/bot_vis/gl_drawing_area.c:7:10: fatal error: 'gdk/gdkx.h' file not found
#include <gdk/gdkx.h>
         ^
1 error generated.
```

No such header file seems to be present in my usr/local/Cellar/gtk+ include folder. I found similar issues online with macports involving gtk and xquartz, but no solution.  Any suggestions to resolve this error?

RussTedrake, 2015-07-10:
Unfortunately, it's not our code, but a tool that we were using that fails:
https://github.com/RobotLocomotion/libbot/issues/3

This is only needed for the 3D visualization tool which we ship.  And we actually have a much newer better one that we are working to make public.  If you can get by without that visualizer, then a work-around would be to edit `drake-distro/tobuild.txt`and comment out the line calling `libbot`.

vikrantvm, 2015-07-10:
Ah..I've just started following your course on edx, so I'm not very sure how much I would need those visualization tools. I will build drake as per your suggestion, and will post a solution, if I find one. Thanks!

RussTedrake, 2015-07-23:
fwiw - @tkoolen posted a solution in https://github.com/RobotLocomotion/libbot/issues/3

closed 2015-07-23 by RussTedrake