RobotLocomotion/drake #1152: Running the BouncingBall example and visualizing the output in 3D

edowson, 2015-07-10:
How can I run the BouncingBall example and visualize its output in 3D?

RussTedrake, 2015-07-10:
The ball example is a simple example of using the matlab-only interface.  It doesn't actually define the 3D geometry, so you can't draw it in 3D.  

Perhaps you'd be interested instead in running the `fallingBricks` and/or `fallingCapsules` demos in `systems/plants/test`

edowson, 2015-07-10:
I'd like to learn how to run this example within the Matlab interface, and just get the simulation output values.

I tried 
sys = BallFlightPhasePlant;
traj = simulate(sys, [0 10], 0.99);

but I guess the matrix has been defined improperly, and get a error
Wrong size.  Expected [2  1] but got a [1  1] instead.

This example differs from the SimpleCTExample.m, in that it models 2 continuous states, and the drake user guide didn't give an example of how to simulate the BouncingBall example.

It would be great if I could get some pointers on how to properly initialize a DrakeSystem object (w.r.t the parameters:
      obj = obj@DrakeSystem(...
        2, ... % number of continuous states
        0, ... % number of discrete states
        0, ... % number of inputs
        1, ... % number of outputs
        false, ... % not direct feedthrough
        true); % time invariant

RussTedrake, 2015-07-10:
You need to pass in a 2x1 vector of initial conditions. 

```
traj = simulate(sys, [0 10], [0.99;0]);
```

edowson, 2015-07-10:
What do the [0 10] and [0.99; 0] conditions correspond to?

RussTedrake, 2015-07-10:
the first one is the time interval.  the second are the initial conditions.

you can always type `help simulate`
or use the doxygen documentation:
http://drake002.csail.mit.edu/drake/doxygen/

edowson, 2015-07-11:
I got the simulation running now, and can see a plot of the ball in the flight phase.

sys = BallFlightPhasePlant;
traj = simulate(sys, [0 10], [0.99; 0]);
fnplt(traj);

[image]

How do I get the ball to bounce when it hits a flat plane and plot its motion?

RussTedrake, 2015-07-11:
```
BallPlant2D.run()
```
(it's a static method... take a look)

edowson, 2015-07-11:
Nice, that worked.

[image]

How can I now run the 3D case? In what format does one go about specifying a CoordinateFrame input to BallVisualizer? I get the following error when I try to run the 3D case

b = BallPlant;
v=BallVisualizer(b);

Error using typecheck (line 24)
Wrong type.  Expected CoordinateFrame but got a BallPlant instead.
Error in Visualizer (line 18)
        typecheck(input_frame,'CoordinateFrame');
Error in BallVisualizer (line 5)
      obj = obj@Visualizer(fr);  % number of inputs = 1

RussTedrake, 2015-07-11:
Sorry, but please spend a little more time reading the code/documentation.  If there are places where the documentation needs work, then requests (or better pull requests!) are great.

closed 2015-07-11 by RussTedrake