RobotLocomotion/drake #1154: systems/plants/test/fallingCapsulesTest causes segmentation fault in drake_viewer: fghDrawGeometrySolid () from /opt/local/lib/libglut.3.dylib

edowson, 2015-07-10:
I tried running the fallingCapsulesTest, and it causes the segmentation fault in drake_viewer, running in debug mode:

Program received signal SIGSEGV, Segmentation fault.
0x00000001000830c2 in fghDrawGeometrySolid () from /opt/local/lib/libglut.3.dylib
(gdb) backtrace full
#0  0x00000001000830c2 in fghDrawGeometrySolid () from /opt/local/lib/libglut.3.dylib
No symbol table info available.
#1  0x0000000100084771 in fghSphere () from /opt/local/lib/libglut.3.dylib
No symbol table info available.
#2  0x000000010115d5a2 in Capsule::draw() ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
No symbol table info available.
#3  0x000000010115de0a in LinkGeometry::draw() ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
No symbol table info available.
#4  0x000000010115e101 in Link::draw() ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
No symbol table info available.
#5  0x000000010115c3c3 in my_draw(_BotViewer*, _BotRenderer*) ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
---Type <return> to continue, or q <return> to quit---
No symbol table info available.
#6  0x0000000100051046 in render_scene () from /project/robotics/library/drake-distro/build/lib/libbot2-vis.1.dylib
No symbol table info available.
#7  0x00000001000511fa in on_gl_expose () from /project/robotics/library/drake-distro/build/lib/libbot2-vis.1.dylib
No symbol table info available.
#8  0x000000010020250f in _gtk_marshal_BOOLEAN__BOXED () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#9  0x0000000100add35e in g_closure_invoke () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
#10 0x0000000100af2185 in signal_emit_unlocked_R () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
#11 0x0000000100af2f1d in g_signal_emit_valist () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
#12 0x0000000100af354a in g_signal_emit () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
---Type <return> to continue, or q <return> to quit---
#13 0x00000001002ff9c2 in gtk_widget_event_internal () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#14 0x00000001002004a3 in gtk_main_do_event () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#15 0x0000000100528f67 in _gdk_window_process_updates_recurse () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#16 0x0000000100529961 in gdk_window_process_updates_internal () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#17 0x0000000100529451 in gdk_window_process_all_updates () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#18 0x0000000100529b0d in gdk_window_update_idle () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#19 0x000000010050cd56 in gdk_threads_dispatch () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#20 0x0000000100c49728 in g_main_context_dispatch () from /opt/local/lib/libglib-2.0.0.dylib
---Type <return> to continue, or q <return> to quit---
No symbol table info available.
#21 0x0000000100c49a0b in g_main_context_iterate () from /opt/local/lib/libglib-2.0.0.dylib
No symbol table info available.
#22 0x0000000100c49c55 in g_main_loop_run () from /opt/local/lib/libglib-2.0.0.dylib
No symbol table info available.
#23 0x00000001001ffddd in gtk_main () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#24 0x000000010000cc23 in main (argc=1, argv=0x7fff5fbff688)
    at /project/robotics/library/drake-distro/drake/systems/plants/viewer/main.cpp:57
        viewer = 0x101817090
        eye = {0, -4, 2}
        lookat = {0, 0, 0}
        fname = 0x10000cf12 ".viewer-prefs"
        viewer_title = {static npos = <optimized out>, 
          _M_dataplus = {<allocator<char>> = {<new_allocator<char>> = {<No data fields>}, <No data fields>}, 
---Type <return> to continue, or q <return> to quit---
            _M_p = 0x1021040c8 "Drake Viewer"}}
        lcm = 0x10210e8f0
        pwidget = 0x10386beb0
        up = {0, 0, 1}

RussTedrake, 2015-07-11:
that definitely shouldn't happen. will try to reproduce it here.

RussTedrake, 2015-07-15:
i assume this was also related to the bad build in #1168.

closed 2015-07-15 by RussTedrake

edowson, 2015-07-15:
Hi Russ, no this was not related to the bad build. I tried it again, and the drake_viewer did not display anything on screen. The fallingBricks test runs but the fallingCapsulesTest do not display anything in the drake_viewer.

RussTedrake, 2015-07-15:
did it crash?

edowson, 2015-07-15:
If you run it normally, drake_viewer doesn't crash. If you use gdb to launch drake_viewer, and run MATLAB R2015a a segmentation fault occurs.

Here is the output from MATLAB:
fallingCapsulesTest
Cannot find required pod snopt
 Added the lcm jar to your javaclasspath (found via cmake)
Warning: Currently all of your LCM traffic will be broadcast to the network, because:
  Route to multicast channel does not run through the loopback interface.
  Consider running: sudo /project/robotics/library/drake-distro/drake/util/setup_loopback_multicast.sh
 
In checkDependency (line 86)
  In BotVisualizer (line 25)
  In RigidBodyManipulator/constructVisualizer (line 1633)
  In TimeSteppingRigidBodyManipulator/constructVisualizer (line 1068)
  In fallingCapsulesTest (line 18) 
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.

Here is the output from gdb:

Program received signal SIGSEGV, Segmentation fault.
0x00000001000830c2 in fghDrawGeometrySolid () from /opt/local/lib/libglut.3.dylib
(gdb) backtrace full
#0  0x00000001000830c2 in fghDrawGeometrySolid () from /opt/local/lib/libglut.3.dylib
No symbol table info available.
#1  0x0000000100084771 in fghSphere () from /opt/local/lib/libglut.3.dylib
No symbol table info available.
#2  0x000000010115d5a2 in Capsule::draw() ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
No symbol table info available.
#3  0x000000010115de0a in LinkGeometry::draw() ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
No symbol table info available.
#4  0x000000010115e101 in Link::draw() ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
No symbol table info available.
#5  0x000000010115c3c3 in my_draw(_BotViewer*, _BotRenderer*) ()
   from /project/robotics/library/drake-distro/build/lib/libdrake_urdf_renderer.dylib
---Type <return> to continue, or q <return> to quit---
No symbol table info available.
#6  0x0000000100051046 in render_scene () from /project/robotics/library/drake-distro/build/lib/libbot2-vis.1.dylib
No symbol table info available.
#7  0x00000001000511fa in on_gl_expose () from /project/robotics/library/drake-distro/build/lib/libbot2-vis.1.dylib
No symbol table info available.
#8  0x000000010020250f in _gtk_marshal_BOOLEAN__BOXED () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#9  0x0000000100add35e in g_closure_invoke () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
#10 0x0000000100af2185 in signal_emit_unlocked_R () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
#11 0x0000000100af2f1d in g_signal_emit_valist () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
#12 0x0000000100af354a in g_signal_emit () from /opt/local/lib/libgobject-2.0.0.dylib
No symbol table info available.
---Type <return> to continue, or q <return> to quit---
#13 0x00000001002ff9c2 in gtk_widget_event_internal () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#14 0x00000001002004a3 in gtk_main_do_event () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#15 0x0000000100528f67 in _gdk_window_process_updates_recurse () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#16 0x0000000100529961 in gdk_window_process_updates_internal () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#17 0x0000000100529451 in gdk_window_process_all_updates () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#18 0x0000000100529b0d in gdk_window_update_idle () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#19 0x000000010050cd56 in gdk_threads_dispatch () from /opt/local/lib/libgdk-x11-2.0.0.dylib
No symbol table info available.
#20 0x0000000100c49728 in g_main_context_dispatch () from /opt/local/lib/libglib-2.0.0.dylib
---Type <return> to continue, or q <return> to quit---
No symbol table info available.
#21 0x0000000100c49a0b in g_main_context_iterate () from /opt/local/lib/libglib-2.0.0.dylib
No symbol table info available.
#22 0x0000000100c49c55 in g_main_loop_run () from /opt/local/lib/libglib-2.0.0.dylib
No symbol table info available.
#23 0x00000001001ffddd in gtk_main () from /opt/local/lib/libgtk-x11-2.0.0.dylib
No symbol table info available.
#24 0x000000010000cc23 in main (argc=1, argv=0x7fff5fbff650)
    at /project/robotics/library/drake-distro/drake/systems/plants/viewer/main.cpp:57
        viewer = 0x103805090
        eye = {0, -4, 2}
        lookat = {0, 0, 0}
        fname = 0x10000cf12 ".viewer-prefs"
        viewer_title = {static npos = <optimized out>, 
          _M_dataplus = {<allocator<char>> = {<new_allocator<char>> = {<No data fields>}, <No data fields>}, 
---Type <return> to continue, or q <return> to quit---
            _M_p = 0x101717a28 "Drake Viewer"}}
        lcm = 0x10200f0a0
        pwidget = 0x103046eb0
        up = {0, 0, 1}

edowson, 2015-07-15:
Hi Russ, perhaps we should reopen this issue?

If you can send me a link to an ftp dropsite, I can zip and send across my current drake-distro rigidbody branch, with all the binary files, so that you can try to replicate the issue at your end. The total size of the zip archive is around 500MB.