RobotLocomotion/drake #1165: update matlab and visual studio minimum version requirements on windows (so we have full c++11 support)

RussTedrake, 2015-07-13:


tkoolen, 2015-07-13:
MSVC/Matlab compatibility:
* http://www.mathworks.com/support/compilers/R2012a/win64.html
* http://www.mathworks.com/support/compilers/R2012b/win64.html
* http://www.mathworks.com/support/compilers/R2013a/
* http://www.mathworks.com/support/compilers/R2013b/
* http://www.mathworks.com/support/compilers/R2014a/
* http://www.mathworks.com/support/compilers/R2014b/

Short story: MSVC 2012 is supported from Matlab 2013a onwards, MSVC 2013 is supported from Matlab 2014a onwards.

C++11 compatibility for MSVC:
https://msdn.microsoft.com/en-us/library/hh567368.aspx

Even MSVC 2013 is not very up to date with C++11, but it's much better than 2010. I would personally like MSVC 2013 because 1) compared to MSVC 2012, it does defaulted and deleted functions, delegating constructors, and variadic templates 2) I imagine if you have access to Matlab 2013a, you probably also have access to 2014a

RussTedrake, 2015-07-13:
is msvc2013 freely available?

tkoolen, 2015-07-13:
Looks like there's a free version called Visual Studio Community 2013:
https://www.visualstudio.com/en-us/products/visual-studio-community-vs.aspx

> An unlimited number of users within an organization can use Visual Studio Community for the following scenarios: in a classroom learning environment, for academic research, or for contributing to open source projects.

RussTedrake, 2015-08-23:
note that gcc4.7 is a little annoying to install on ubuntu 12.04, but possible:
http://askubuntu.com/a/315220

rdeits, 2015-08-23:
That's basically what I'm already doing in the Travis gcc-4.8 branch (since
Travis runs 12.04)

RussTedrake, 2015-08-27:
At this point, we have officially made this change.  I have updated the compiler minimal version logic for gcc in the drake CMakeLists.txt, but we still need to do it for MSVC.  Minimal version is now MSVC 2013.

tkoolen, 2015-08-27:
Awesome!

RussTedrake, 2015-10-07:
According to this, we should increment our minimal supported version of MATLAB to R2014a.

tkoolen, 2015-10-07:
Yeah, probably a good idea. But the fact that Matlab 2012a worked on drake003 for so long makes me wonder if we really need to stop supporting older versions. Or maybe we can support older versions for the binary install?

psiorx, 2015-10-07:
I've been testing the Windows precompiled distro on 2013b.

tkoolen, 2015-12-15:
Close?

closed 2015-12-15 by RussTedrake