RobotLocomotion/drake #1168: Pendulum examples crashing MATLAB R2015a on Mac OS X 10.10.4

edowson, 2015-07-13:
Several pendulum examples consistently crash MATLAB R2015a, resulting in an abnormal termination.
- examples/Pendulum/PendulumEnergyControl.m (run by invoking the static run method)
- runDLQR
- runLQR
- runPassive
- runSwingUp

There are a bunch of other examples in the folder, and it looks like nearly all of them are crashing MATLAB R2015a on Mac OS X 10.10.4 consistently with the following error message:

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 000000010a1d5a00  RBX = 00007ff7ca96abc0
  RCX = 0000000000000050  RDX = 0000000000000000
  RSP = 00007fff93f46e7f  RBP = 0000000110794230
  RSI = 0000000102e0d000  RDI = 00000000000006aa

   R8 = 00007ff7ca900000   R9 = 00007ff700000002
  R10 = 000000010a1d5a00  R11 = 00007ff7ca96ab70
  R12 = 00000000000000d0  R13 = 0000000000000000
  R14 = 0000000110794200  R15 = 00000001086e2c21

  RIP = 0000000000000000  RFL = 0000000000003fff

   CS = 0000000000003fff   FS = 0000000000000001   GS = 00000001107942f0

Stack Trace (from fault):
[  0] 0x0000000103424964 /Applications/MATLAB_R2015a.app/bin/maci64/libmwfl.dylib+00031076 _ZN2fl4diag15stacktrace_base7captureERKNS0_14thread_contextEm+00000052
[  1] 0x000000010342802b /Applications/MATLAB_R2015a.app/bin/maci64/libmwfl.dylib+00045099 _ZN2fl4test17terminate_handledEv+00000763
[  2] 0x0000000103427ac7 /Applications/MATLAB_R2015a.app/bin/maci64/libmwfl.dylib+00043719 _ZN2fl4diag13terminate_logEPKcPK17__darwin_ucontext+00000119
[  3] 0x0000000102f55f9b /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcr.dylib+00380827 _Z32mnRunPathDependentInitializationv+00002875
[  4] 0x0000000102f56217 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcr.dylib+00381463 _Z32mnRunPathDependentInitializationv+00003511
[  5] 0x0000000102f543a6 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcr.dylib+00373670 mnFatalSignalHandler+00000262
[  6] 0x00007fff957a2f1a           /usr/lib/system/libsystem_platform.dylib+00020250 _sigtramp+00000026
[  7] 0x0000000102e0d000                                   <unknown-module>+00000000
[  8] 0x00000001086e2c21 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+03550241 _ZN11inPredictor14MaybeVoidCheckEP13Mfh_MATLAB_fniiPK7mpsType16inPredictorFlags+00000033
[  9] 0x00000001086e39e4 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+03553764 _ZN11inPredictor6LookupEP13Mfh_MATLAB_fn18inPredictorContextiP7mpsTypeiPKS3_P16inPredictorFlagsb+00001604
[ 10] 0x00000001086e912c /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+03576108 _Z17inPredictorLookupP13Mfh_MATLAB_fn18inPredictorContextiP7mpsTypeiPKS2_P16inPredictorFlags+00000156
[ 11] 0x00000001086f0fe7 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+03608551 _ZN8Resolver15HandleBadResultEiiP10_m_operandP7mpsTypeiS1_+00000647
[ 12] 0x00000001086f332d /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+03617581 _ZN8Resolver13CallMFunctionEiiP10_m_operandP17m_operand_storageiS1_S3_Pi+00002589
[ 13] 0x00000001086f3b07 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+03619591 _Z22inResolveMFunctionCallP16_m_function_desciiP10_m_operandP17m_operand_storageiS2_S4_PiP13inMarshalTypeiPK19mpsTypeSequenceNlhsPFP11mxArray_tagiE+00000231
[ 14] 0x00000001085622bd /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01974973 _ZN9accelImpl13MFunctionCallEPP8_accelOp+00000365
[ 15] 0x00000001085628e4 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01976548 _ZN9accelImpl4ExecEv+00000180
[ 16] 0x000000010856264e /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01975886 _ZNK9accelCode4CallEP13inMarshalTypePi+00000094
[ 17] 0x00000001086a84f6 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+03310838 _ZN5inJit17ExecuteHotSegmentEP15_inJitAccelInfoP7opcodesPiPl+00003894
[ 18] 0x00000001084a0764 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01181540 _Z26inExecuteMFunctionOrScriptP6Mfh_mpb+00000596
[ 19] 0x0000000108536819 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01796121 _Z10inRunMfileiPP11mxArray_tagiS1_P6Mfh_mpP15inWorkSpace_tag+00003401
[ 20] 0x00000001080fe4bd /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_dispatcher.dylib+00386237 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00000621
[ 21] 0x00000001080fe5f3 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_dispatcher.dylib+00386547 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000035
[ 22] 0x000000011145cf04 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcos_impl.dylib+01027844 _ZNK4mcos17MScopedMethodInfo6invokeEPKNS_9COSClientERiPNS_8COSValueEiPKS5_+00000548
[ 23] 0x00000001113fcb77 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcos_impl.dylib+00633719 _ZN24OpaqueMethodCallMetaData11callMMethodEPN4mcos13COSMethodInfoEPP11mxArray_tagPKNS0_9COSClientE+00000359
[ 24] 0x00000001113ff643 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcos_impl.dylib+00644675 _ZN24OpaqueMethodCallMetaData23callMethodOrConstructorEP11mxArray_tagbiPS1_iPPKS0_bPKN4mcos9COSClientEPS9_PPNS6_13COSMethodInfoE+00000259
[ 25] 0x000000011140014a /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcos_impl.dylib+00647498 _ZN20OpaqueMethodMetaData23callMethodOrConstructorEP11mxArray_tagbiPS1_iPPKS0_bPKN4mcos9COSClientEPS9_PPNS6_13COSMethodInfoE+00000378
[ 26] 0x00000001113fea94 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcos_impl.dylib+00641684 _Z22omCallMethodWithClientP11mxArray_tagbS0_iPS0_iPPKS_PKN4mcos9COSClientEPS8_PPNS5_13COSMethodInfoE+00000340
[ 27] 0x000000011148a448 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcos_impl.dylib+01213512 _Z18omCallStaticMethodP11mxArray_tagiPS0_iPPKS_+00000088
[ 28] 0x00000001084ce27d /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01368701 _Z17inFullClassDotFcniPP11mxArray_tagiS1_+00000749
[ 29] 0x00000001080bc936 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_dispatcher.dylib+00117046 _ZN11Mfh_builtin11dispatch_mfEiPP11mxArray_tagiS2_+00000086
[ 30] 0x00000001080a4cdf /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_dispatcher.dylib+00019679 _ZN13Mfh_MATLAB_fn11dispatch_fhEiPP11mxArray_tagiS2_+00000335
[ 31] 0x00000001084fb03d /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01552445 _Z19inDispatchFromStackiPKcii+00001581
[ 32] 0x00000001084a6698 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01205912 inPcodeWalker+00019048
[ 33] 0x00000001084a4377 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01196919 inPcodeWalker+00010055
[ 34] 0x000000010849f76b /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01177451 inInterPcode+00000875
[ 35] 0x00000001084a0886 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01181830 _Z26inExecuteMFunctionOrScriptP6Mfh_mpb+00000886
[ 36] 0x0000000108536819 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01796121 _Z10inRunMfileiPP11mxArray_tagiS1_P6Mfh_mpP15inWorkSpace_tag+00003401
[ 37] 0x00000001080fe4f2 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_dispatcher.dylib+00386290 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00000674
[ 38] 0x00000001080fe5f3 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_dispatcher.dylib+00386547 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000035
[ 39] 0x00000001084ddebd /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01433277 _Z23inEvalPcodeHeaderToWordP15_memory_contextiPP11mxArray_tagP12_pcodeheaderP6Mfh_mpj+00000141
[ 40] 0x0000000108498adb /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01149659 _Z25in_local_call_with_setjmpIN5boost3_bi6bind_tIvPFvP15_memory_contextPiPP11mxArray_tagP12_pcodeheaderjENS1_5list5INS1_5valueIS4_EENS0_3argILi1EEENSG_ILi2EEENSE_ISA_EENSE_IiEEEEEEEN6mlutil14cmddistributor17inExecutionStatusET_S5_S8_b+00000171
[ 41] 0x0000000108493dc0 /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01129920 _Z12inEvalStringP15_memory_contextRKNSt3__112basic_stringIDsNS1_11char_traitsIDsEENS1_9allocatorIDsEEEE8EvalTypeiPP11mxArray_tag12inDebugCheckP12_pcodeheaderPib+00002608
[ 42] 0x00000001084943dd /Applications/MATLAB_R2015a.app/bin/maci64/libmwm_interpreter.dylib+01131485 _ZN12InterpBridge22EvalCmdWithLocalReturnERKNSt3__112basic_stringIDsNS0_11char_traitsIDsEENS0_9allocatorIDsEEEEPibb+00000141
[... cut ...]
[ 57] 0x0000000102f16a6a /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcr.dylib+00121450 _ZN3mcr7runtime17InterpreterThread4Impl3runERKN5boost10shared_ptrIS2_EEPNS2_12init_contextE+00000346
[ 58] 0x0000000102eff156 /Applications/MATLAB_R2015a.app/bin/maci64/libmwmcr.dylib+00024918 _Z26run_init_and_handle_eventsPv+00000054
[ 59] 0x00007fff95797268            /usr/lib/system/libsystem_pthread.dylib+00012904 _pthread_body+00000131
[ 60] 0x00007fff957971e5            /usr/lib/system/libsystem_pthread.dylib+00012773 _pthread_body+00000000
[ 61] 0x00007fff9579541d            /usr/lib/system/libsystem_pthread.dylib+00005149 thread_start+00000013

RussTedrake, 2015-07-13:
I have not seen this (nor your visualizer crash in #1154) on any of our other systems, and I don't see anything drake related in your stack trace.

edowson, 2015-07-13:
The crash in the stack trace is purely related to MATLAB R2015a, I agree. Some of the other basic examples such as the Acrobot work as expected. I'm surprised that the basic Pendulum example crashes MATLAB R2015a. The problem is consistently reproducible at my end, including switching between a working session with the Acrobot, QuadCopter, BouncingBall, and completely crashing immediately while running the Pendulum example.

RussTedrake, 2015-07-13:
To be fair, I have not tested with 2015a yet on mac 10.10.4.  But those examples are pretty robust.

edowson, 2015-07-13:
Rebuilding the entire drake-distro rigidbody branch, after manually switching drake to the HEAD of the master branch, helped get past this issue. Perhaps there were some artifacts left over from a previous build. I didn't issue a make clean command, I just deleted the build folder. I guess mex files from a previous build were present that was out of sync with the drake matlab sources which caused the crash.

It would be good to update the drake-distro rigidbody branch's drake submodule to the current HEAD of the master branch.

closed 2015-07-13 by edowson