RobotLocomotion/drake #1170: gcc-4.8 build error on Mac OS X 10.10.4 with Gurobi 6.0.4 for systems/controllers/QPCommon.cpp:925:30: error: 'isnan' was not declared in this scope

edowson, 2015-07-13:
I get the a build error while attempting to build drake-distro using rigidbody branch, but with the gurobi pod inserted into the build. I have installed gurobi-6.0.4.

The build process fails with the following error:

[ 99%] Built target constructQPDataPointerMex
[100%] Building CXX object systems/controllers/CMakeFiles/drakeQPCommon.dir/QPCommon.cpp.o
/project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp: In function 'int setupAndSolveQP(NewQPControllerData*, std::shared_ptr<drake::lcmt_qp_controller_input>, DrakeRobotState&, const Eigen::Ref<Eigen::Matrix<bool, -1, 1> >&, QPControllerOutput*, std::shared_ptr<QPControllerDebugData>)':
/project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp:925:30: error: 'isnan' was not declared in this scope
     if (isnan(qp_output->u(i))) qp_output->u(i) = 0;
                              ^
/project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp:925:30: note: suggested alternative:
In file included from /opt/local/include/gcc48/c++/complex:44:0,
                 from /project/robotics/library/drake-distro/build/include/eigen3/Eigen/Core:28,
                 from /project/robotics/library/drake-distro/build/include/eigen3/Eigen/Dense:1,
                 from /project/robotics/library/drake-distro/drake/systems/controllers/controlUtil.h:9,
                 from /project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.h:4,
                 from /project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp:1:
/opt/local/include/gcc48/c++/cmath:632:5: note:   'std::isnan'
     isnan(_Tp __x)
     ^
systems/controllers/CMakeFiles/drakeQPCommon.dir/build.make:54: recipe for target 'systems/controllers/CMakeFiles/drakeQPCommon.dir/QPCommon.cpp.o' failed
gmake[4]: *** [systems/controllers/CMakeFiles/drakeQPCommon.dir/QPCommon.cpp.o] Error 1
CMakeFiles/Makefile2:5727: recipe for target 'systems/controllers/CMakeFiles/drakeQPCommon.dir/all' failed
gmake[3]: *** [systems/controllers/CMakeFiles/drakeQPCommon.dir/all] Error 2
Makefile:126: recipe for target 'all' failed
gmake[2]: *** [all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

edowson, 2015-07-13:
ah, I think it's probably related to the fact that I'm working off the rigidbody branch, and the drake-distro .gitmodules would probably be set to an old revisions that is 674 commits behind drake master

$ cd drake-distro/drake
$ git checkout master
M	examples/BouncingBall/BallVisualizer2D.m
Previous HEAD position was 57fb6ae... Merge pull request #1127 from RussTedrake/cygwin_check
Switched to branch 'master'
Your branch is behind 'origin/master' by 674 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
$ git pull

I'll do another rebuild and check. Perhaps this could be the reason for the inconsistencies and crashes that I'm seeing on my system.

It would be good to update the drake-distro rigibbody branch to fetch the latest changes for the submodules that it contains.

edowson, 2015-07-13:
I updated my drake submodule with the latest changes and performed a clean rebuild with gurobi-6.0.4, and the gcc-4.8 error still persists.

[ 93%] Built target drakeSide
[ 93%] Building CXX object systems/controllers/CMakeFiles/drakeQPCommon.dir/QPCommon.cpp.o
/project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp: In function 'int setupAndSolveQP(NewQPControllerData*, std::shared_ptr<drake::lcmt_qp_controller_input>, DrakeRobotState&, const Eigen::Ref<Eigen::Matrix<bool, -1, 1> >&, const std::map<Side, ForceTorqueMeasurement>&, QPControllerOutput*, std::shared_ptr<QPControllerDebugData>)':
/project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp:917:30: error: 'isnan' was not declared in this scope
     if (isnan(qp_output->u(i))) qp_output->u(i) = 0;
                              ^
/project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp:917:30: note: suggested alternative:
In file included from /opt/local/include/gcc48/c++/complex:44:0,
                 from /project/robotics/library/drake-distro/build/include/eigen3/Eigen/Core:28,
                 from /project/robotics/library/drake-distro/build/include/eigen3/Eigen/Dense:1,
                 from /project/robotics/library/drake-distro/drake/systems/controllers/controlUtil.h:8,
                 from /project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.h:4,
                 from /project/robotics/library/drake-distro/drake/systems/controllers/QPCommon.cpp:1:
/opt/local/include/gcc48/c++/cmath:632:5: note:   'std::isnan'
     isnan(_Tp __x)
     ^
systems/controllers/CMakeFiles/drakeQPCommon.dir/build.make:54: recipe for target 'systems/controllers/CMakeFiles/drakeQPCommon.dir/QPCommon.cpp.o' failed
gmake[4]: *** [systems/controllers/CMakeFiles/drakeQPCommon.dir/QPCommon.cpp.o] Error 1
CMakeFiles/Makefile2:6221: recipe for target 'systems/controllers/CMakeFiles/drakeQPCommon.dir/all' failed
gmake[3]: *** [systems/controllers/CMakeFiles/drakeQPCommon.dir/all] Error 2
Makefile:126: recipe for target 'all' failed
gmake[2]: *** [all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

RussTedrake, 2015-07-13:
@psiorx - this might be our old isnan friend caught by a new compiler version.  Can u take a quick look?

edowson, 2015-07-13:
The following patch fixes this issue:

diff --git a/systems/controllers/QPCommon.cpp b/systems/controllers/QPCommon.cpp
index c485220..c3f9355 100644
--- a/systems/controllers/QPCommon.cpp
+++ b/systems/controllers/QPCommon.cpp
@@ -3,6 +3,7 @@
 #include "controlUtil.h"
 #include <map>
 #include <memory>
+#include <cmath>
 #include <lcm/lcm-cpp.hpp>
 #include "lcmUtil.h"
 #include "testUtil.h"
@@ -914,7 +915,7 @@ int setupAndSolveQP(
   // use transpose because B_act is orthogonal
   qp_output->u = pdata->B_act.transpose()*(pdata->H_act*qp_output->qdd + pdata->C_act - D_act*beta);
   for (int i=0; i < qp_output->u.size(); i++) {
-    if (isnan(qp_output->u(i))) qp_output->u(i) = 0;
+      if (std::isnan(qp_output->u(i))) qp_output->u(i) = 0;
   }
   //y = pdata->B_act.jacobiSvd(ComputeThinU|ComputeThinV).solve(pdata->H_act*qdd + pdata->C_act - Jz_act.transpose()*lambda - D_act*beta);

closed 2015-07-13 by edowson