RobotLocomotion/drake #1172: compiler warning on drakeShapes

RussTedrake, 2015-07-14:
just saw this one.  @avalenzu - can you take a look?

```
[ 59%] Building CXX object systems/plants/shapes/CMakeFiles/drakeShapes.dir/Geometry.cpp.o
/tmp/drake-master/build/include/eigen3/Eigen/src/Core/CommaInitializer.h: In member function 'virtual void DrakeShapes::MeshPoints::getBoundingBoxPoints(Eigen::Matrix3Xd&) const':
/tmp/drake-master/build/include/eigen3/Eigen/src/Core/CommaInitializer.h:71:5: warning: '*((void*)& min_pos +24)' may be used uninitialized in this function [-Wuninitialized]
/tmp/drake-master/drake-19757/systems/plants/shapes/Geometry.cpp:285:14: note: '*((void*)& min_pos +24)' was declared here
/tmp/drake-master/build/include/eigen3/Eigen/src/Core/CommaInitializer.h:71:5: warning: '*((void*)& max_pos +24)' may be used uninitialized in this function [-Wuninitialized]
/tmp/drake-master/drake-19757/systems/plants/shapes/Geometry.cpp:286:14: note: '*((void*)& max_pos +24)' was declared here
/tmp/drake-master/build/include/eigen3/Eigen/src/Core/CommaInitializer.h: In member function 'virtual void DrakeShapes::Mesh::getBoundingBoxPoints(Eigen::Matrix3Xd&) const':
/tmp/drake-master/build/include/eigen3/Eigen/src/Core/CommaInitializer.h:71:5: warning: '*((void*)& min_pos +24)' may be used uninitialized in this function [-Wuninitialized]
/tmp/drake-master/drake-19757/systems/plants/shapes/Geometry.cpp:260:16: note: '*((void*)& min_pos +24)' was declared here
/tmp/drake-master/build/include/eigen3/Eigen/src/Core/CommaInitializer.h:71:5: warning: '*((void*)& max_pos +24)' may be used uninitialized in this function [-Wuninitialized]
/tmp/drake-master/drake-19757/systems/plants/shapes/Geometry.cpp:261:16: note: '*((void*)& max_pos +24)' was declared here
```

avalenzu, 2015-07-14:
Will do.

avalenzu, 2015-07-14:
Russ, can you give more details on what system and/or compiler flags you used to get this warning? I haven't been able to reproduce it.

PR #1173 (2015-07-14): Fix off-by one error in Geometry.cpp.
Resolves #1172. Also we must not be calling the bounding-box C++ functions
because this would have caused a segfault.
commit 9e0093d6 (2015-07-14): Fix off-by one error in Geometry.cpp.

Resolves #1172. Also we must not be calling the bounding-box C++ functions
because this would have caused a segfault.
closed 2015-07-14 by RussTedrake