RobotLocomotion/drake #1175: make user wiki page for contributed examples/discussion, etc.

RussTedrake, 2015-07-14:
we use this github issue tracker for a lot of things, but want to have a place for user contributed examples, etc.  

recommendations for wiki / tools would be appreciated!

RussTedrake, 2015-08-05:
A little more on this -- we're thinking this will be an external website.  (because the drake wiki has limited permissions options).  It might be more of a discussion forum than a wiki, really.  what tools do similar sites use?  anybody have favorite examples?

psiorx, 2015-08-05:
An external site with discussion forum functionality would give us more control over user contributed content.  The GitHub wiki permissions system isn't very flexible.

https://www.phpbb.com/about/ seems like a good open source platform for doing something like this but I'm sure there are many alternatives.

Any hosting ideas?  Maybe it could temporarily live on one of the build servers while under development.

psiorx, 2015-08-05:
After evaluating a few open source options for this, I've settled on myBB.  I got a prototype of this working today using an apache/mysql backend running locally on my machine.  The look and feel can be customized heavily using CSS so this is just an example.  Here's a screenshot of what I have so far:

[image]

RussTedrake, 2015-08-05:
Cool.  Did you also consider services that are hosted somewhere else (so we don’t have to maintain yet another server?)

psiorx, 2015-08-06:
I hadn't looked at it yesterday but I will today.  We could also throw up an AWS instance (though that doesn't completely solve the server maintenance problem).

I found a few options for free hosted forums:
http://www.forumer.com/
http://www.forumotion.com/
http://www.lefora.com/

RussTedrake, 2015-12-15:
recommendation:  askbot or maybe even piazza?

sheim, 2016-01-11:
I concur with askbot. Makes searching through past questions/answers a lot easier, and I think the familiar feel can encourage people to ask/answer "silly" questions (I know I hesitate to ask things here. You guys are super friendly, but MIT is still a pretty daunting name ;) ).

RussTedrake, 2016-01-11:
That’s a really good point (one I hadn’t considered).  Thnks!

jwnimmer-tri, 2016-03-03:
In other communities, I have seen using stackoverflow.com work very well, with (in our case) a +drake tag to support filtering.  You can sign up to get an email anytime someone posts a new +drake question / suggestion there.

RussTedrake, 2016-05-01:
@nkoenig, @gerkey -- would be super interested in your thoughts on this?

gerkey, 2016-05-02:
For Q&A forums, we use askbot extensively:

* http://answers.ros.org
* http://answers.gazebosim.org

It works great. A little work is required to keep down the spam. We work with the main developer of askbot for help with administrative and maintenance tasks on those servers.

For wiki/tutorial-like functionality, we've used many things:

* MoinMoin (http://wiki.ros.org): not recommended for new systems for reasons of scaling and spam control; we're sticking with that tool only because it will take a huge effort to migrate to something else
* GitHub wiki (https://github.com/ros2/ros2/wiki): works well, but, as already mentioned, the permissions model is not very flexible
* Custom tooling to present source-controlled tutorial content (code from https://bitbucket.org/osrf/gazebo_tutorials/src is presented at http://gazebosim.org/tutorials; code from https://github.com/ros2/design is presented at http://design.ros2.org/): adds some maintenance because we're doing it ourselves, but people can contribute via pull requests, which is good

In general, it's hard to find a good compromise across competing goals: lower the barrier to entry to make edits, combat spam, customize the look and feel of the presented material, minimize maintenance effort, make edits available right away, review edits before accepting them, allow free-form submission of content, enforce structure and flow. The gold standard is obviously wikipedia, but they rely on an army of editors who keep an eye on everything, which is hard to do for a smaller project.

david-german-tri, 2016-05-25:
Reassigned to RLG and @psiorx per discussion with @RussTedrake

RussTedrake, 2018-11-07:
We've settled with stackoverflow and a gallery for now.  Can re-open if we decide to revisit this.

closed 2018-11-07 by RussTedrake