RobotLocomotion/drake #1179: implement TaylorVar in c++

RussTedrake, 2015-07-15:
Again @tkoolen has done a beautiful first pass at this, using Eigen's autodiff scalar
https://www.dropbox.com/s/24lrdb5lgmw7x2e/AutoDiffTest.tar?dl=0

tkoolen, 2015-07-17:
Can I try using AutoDiffScalar in the manipulatorDynamics function of AcrobotPlant? Or are you already working on this?

RussTedrake, 2015-07-17:
Haven'f started.  By all means go for it.  Thanks!

tkoolen, 2015-07-17:
Just got an initial implementation working. See https://github.com/tkoolen/drake/tree/tk-autodiff-scalar.

Still to do:
1. move utility functions to a better place (I just put them in manipulatorDynamics.cpp for now)
2. add support for higher order TaylorVars (non-trivial)

I think I'll skip 2 for now, do 1, and submit a pull request.

tkoolen, 2015-07-17:
OK, I did 1. I also added a templated function called eigenToMatlabGeneral, with specializations that call eigenToMatlab, eigenToTrigPoly, or eigenToTaylorVar based on the type of the input. This allowed me to clean manipulatorDynamics.cpp up quite a bit. Have a look.

tkoolen, 2015-07-18:
See https://github.com/RobotLocomotion/drake/pull/1191.

RussTedrake, 2015-07-18:
excellent!

RussTedrake, 2015-08-27:
resolving as duplicate of #1199.

closed 2015-08-27 by RussTedrake