RobotLocomotion/drake #1184: Trajectories through walls in with office.urdf

peteflorence, 2015-07-16:
@scottviteri @rdeits 

With the current office.urdf from this May 8 commit https://github.com/RobotLocomotion/drake/commit/25ea55e441d63b8d17557be6a720bb6486f6ab53, runMixedIntegerOffice gives me a trajectory that goes through this wall:

[image]

I reverted to the office.urdf version from before that commit https://github.com/RobotLocomotion/drake/blob/67f43ec7d4eb4c83ad1a258ede1724de2de09705/examples/Quadrotor/office.urdf, and it gave me a collision-free trajectory:

[image]

@scottviteri should we not just revert to your older version?

scottviteri, 2015-07-16:
> With the current office.urdf from this May 8 commit 25ea55e

For right this second you can revert to the old version, but I'll fix the
new version today.

-Scott

RussTedrake, 2015-08-27:
resolved in  #1187

closed 2015-08-27 by RussTedrake