RobotLocomotion/drake #1196: Director crashing

peteflorence, 2015-07-21:
Crashes while trying to launch from MATLAB.  Error message below.

Was all working for me a couple days ago, but after a new git pull is crashing on me.

Drake_viewer is working however (if I go into drake/pod-build/bin and run ./drake_viewer)

 

Process:               ddConsoleApp [47705]
Path:                  /Users/USER/*/ddConsoleApp
Identifier:            ddConsoleApp
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Terminal [433]
User ID:               502

Date/Time:             2015-07-21 16:55:18.077 -0400
OS Version:            Mac OS X 10.10.4 (14E46)
Report Version:        11
Anonymous UUID:        2A709556-CF86-948D-8926-0EB7812E1496

Time Awake Since Boot: 4200 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff90bd0286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff87554b53 abort + 129
2   QtCore                        	0x000000010c5c69aa qt_message_output(QtMsgType, char const*) + 203
3   QtCore                        	0x000000010c5c6b57 qt_message(QtMsgType, char const*, __va_list_tag*) + 203
4   QtCore                        	0x000000010c5c6595 qFatal(char const*, ...) + 159
5   QtGui                         	0x000000010ba91a0e QWidgetPrivate::QWidgetPrivate(int) + 790
6   QtGui                         	0x000000010ba92ac8 QWidget::QWidget(QWidget*, QFlags<Qt::WindowType>) + 52
7   libQVTK.5.10.dylib            	0x0000000108a75eec QVTKWidget::QVTKWidget(QWidget*, QFlags<Qt::WindowType>) + 32
8   libddApp.dylib                	0x000000010666de7f ddQVTKWidgetView::ddQVTKWidgetView(QWidget*) + 143
9   libddApp.dylib                	0x000000010665c476 ddPythonQtDecorators::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 182
10  libddApp.dylib                	0x000000010665d728 ddPythonQtDecorators::qt_metacall(QMetaObject::Call, int, void**) + 56
11  libPythonQt.dylib             	0x0000000107dc4ccb PythonQtCallSlot(PythonQtClassInfo*, QObject*, _object*, bool, PythonQtSlotInfo*, void*, _object**, void**) + 939
12  libPythonQt.dylib             	0x0000000107dc7196 PythonQtSlotFunction_CallImpl(PythonQtClassInfo*, QObject*, PythonQtSlotInfo*, _object*, _object*, void*, void**) + 982
13  libPythonQt.dylib             	0x0000000107db76da PythonQtInstanceWrapper_init(PythonQtInstanceWrapperStruct*, _object*, _object*) + 122
14  org.python.python             	0x00000001069c6aab type_call + 182
15  org.python.python             	0x0000000106983ea7 PyObject_Call + 99
16  org.python.python             	0x0000000106a000cb PyEval_EvalFrameEx + 11417
17  org.python.python             	0x00000001069fd200 PyEval_EvalCodeEx + 1387
18  org.python.python             	0x00000001069a1fb5 function_call + 352
19  org.python.python             	0x0000000106983ea7 PyObject_Call + 99
20  org.python.python             	0x000000010698ed32 instancemethod_call + 174
21  org.python.python             	0x0000000106983ea7 PyObject_Call + 99
22  org.python.python             	0x0000000106a0307a PyEval_CallObjectWithKeywords + 93
23  org.python.python             	0x000000010698cf9d PyInstance_New + 126
24  org.python.python             	0x0000000106983ea7 PyObject_Call + 99
25  org.python.python             	0x0000000106a000cb PyEval_EvalFrameEx + 11417
26  org.python.python             	0x0000000106a0391d fast_function + 262
27  org.python.python             	0x0000000106a0079f PyEval_EvalFrameEx + 13165
28  org.python.python             	0x00000001069fd200 PyEval_EvalCodeEx + 1387
29  org.python.python             	0x0000000106a01245 PyEval_EvalFrameEx + 15891
30  org.python.python             	0x00000001069fd200 PyEval_EvalCodeEx + 1387
31  org.python.python             	0x0000000106a0388c fast_function + 117
32  org.python.python             	0x0000000106a0079f PyEval_EvalFrameEx + 13165
33  org.python.python             	0x00000001069fd200 PyEval_EvalCodeEx + 1387
34  org.python.python             	0x00000001069a1fb5 function_call + 352
35  org.python.python             	0x0000000106983ea7 PyObject_Call + 99
36  org.python.python             	0x0000000106a2e26d RunModule + 113
[... 466 lines cut ...]
mapped file                       118.9M
shared memory                         4K
===========                      =======
TOTAL                             666.9M

Model: MacBookPro11,3, BootROM MBP112.0138.B15, 4 processors, Intel Core i7, 2.8 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.5f8 15969, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en4
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM1024F, 1 TB
USB Device: Internal Memory Card Reader
USB Device: 4-Port USB 3.0 Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB2.0 Hub
USB Device: Dell USB Keyboard
USB Device: USB OPTICAL MOUSE
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 1, 5.5

peteflorence, 2015-07-22:
Error message pops up in a "Problem Report" window like this:

[image]

peteflorence, 2015-07-22:
So far I have tried:

- Git cloning a new drake-distro repo and building form scratch

- Launching ddapp from terminal, which gives this error message in the terminal:

30-27-191:bin pflomacpro$ ./ddConsoleApp -m ddapp.drakevisualizer
objc[55165]: Class QCocoaColorPanelDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QMacSoundDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaPanel is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaView is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaWindow is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaWindowDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaMenuLoader is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QNSApplication is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaApplicationDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaWindowCustomThemeFrame is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaToolBarDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaMenu is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QNSOpenSavePanelDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaFontPanelDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QNSPanelProxy is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QNSWindowProxy is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaPageLayoutDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QCocoaPrintPanelDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QNSImageView is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QNSStatusItem is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
objc[55165]: Class QNSMenu is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.
QWidget: Must construct a QApplication before a QPaintDevice
Abort trap: 6

patmarion, 2015-07-22:
@peteflorence it looksl ike you might have two versions of Qt installed.  The homebrew copy is in /usr/lib, but you also seem to have Qt frameworks in /opt/local.  What's up with that?  Did you install that?  Note the error message you get:

Class QCocoaColorPanelDelegate is implemented in both /usr/local/lib/QtGui.framework/Versions/4/QtGui and /opt/local/Library/Frameworks/QtGui.framework/Versions/4/QtGui. One of the two will be used. Which one is undefined.

See it warns about /usr/local clashing with /opt/local.  I'm skeptical of the copy you have in /opt/local.

peteflorence, 2015-07-22:
Thanks for looking into Pat

Yeah I have PyQt installed for some crazyflie stuff.  Let me delete /opt/local library and see what we get

Odd this error never came up before...

peteflorence, 2015-07-22:
Still working on deleting one of my versions of Qt

I tried just deleting the /opt/local/Frameworks but that gave me new a new runtime error below.  Did a make clean and make -j and it's still linking to to the /opt/local Qt library

[ 98%] make[7]: *** No rule to make target `/opt/local/lib/libQtUiTools.dylib', needed by `libPythonQt.dylib'.  Stop.

30-27-191:bin pflomacpro$ ./ddConsoleApp -m ddapp.drakevisualizer
dyld: Library not loaded: /opt/local/Library/Frameworks/QtUiTools.framework/Versions/4/QtUiTools
  Referenced from: /Users/pflomacpro/drake-distro/build/lib/libPythonQt.dylib
  Reason: image not found
Trace/BPT trap: 5

peteflorence, 2015-07-22:
There was a long list of issues related to debugging this, but the main takeaways are:

- (no surprise) Having macports on your machine can cause lots of issues (we ended up doing a full uninstall according to https://guide.macports.org/chunked/installing.macports.uninstalling.html)
- Installing Qt from a dmg or tarball can also cause lots of issues.  The only way we could find to uninstall was by reinstalling (https://download.qt.io/archive/qt/4.8/4.8.6/) and then running the uninstall-qt.py script

Thanks @tkoolen and @psiorx for help

closed 2015-07-22 by peteflorence