RobotLocomotion/drake #1197: Acrobot runMaxROA fails

tkoolen, 2015-07-22:
Going through CDash history for examples/Acrobot/runMaxROA, it looks like this test has not been passing (timing out) in master since June 22. See http://kobol.csail.mit.edu/cd/testSummary.php?project=1&name=examples%2FAcrobot%2FrunMaxROA&date=2015-06-22, you can change the date in the URL.

Currently in master, if it can't find a solution it hits a keyboard statement and waits for input until it times out. I replaced that keyboard statement by an error in https://github.com/RobotLocomotion/drake/pull/1190 so that we're aware something is wrong and we're not needlessly adding to build time. We need to figure out what's wrong though.

tkoolen, 2015-07-22:
I traced this to https://github.com/RobotLocomotion/drake/pull/1116. Disabling input limits (commenting out line 74 of maxROAFeedback.m) makes the test pass again. @psiorx, could you look at this?

psiorx, 2015-07-23:
Seems like SeDuMi runs into numerical issues with the SDP if the input limits are too aggressive.  I changed the input limits from [-4, 4] to [-20, 20] and it works.  We should probably make this the default in Acrobot/runMaxROA.

tkoolen, 2015-07-23:
Sounds good. Thanks for investigating.

RussTedrake, 2015-08-27:
duplicate of #1279 (which has the known issue logic in it, so I will close this one and leave that one open)

closed 2015-08-27 by RussTedrake