RobotLocomotion/drake #1201: ComputeGradients crashes MATLAB

aespielberg, 2015-07-26:
Hi all,

When I try to do use computeGradients as in the example in the documentation:

```
cd examples/Pendulum p=PendulumPlant(); generateGradients('dynamics',3,'dynamicsGradients',p,0,randn(2,1),0);
```

MATLAB prints this:

```
Added the lcm jar to your javaclasspath (found via cmake)
Warning: Currently all of your LCM traffic will be broadcast to the network, because:
  Loopback interface is not set to multicast.
  Consider running: sudo /home/aespielberg/drake/drake-distro/drake/util/setup_loopback_multicast.sh
 
> In checkDependency (line 86)
  In LCMCoordinateFrame (line 13)
  In PendulumInput (line 5)
  In PendulumPlant (line 25) 
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
```

And then I get the dialog box of doom:

"MATLAB has encountered an internal problem and needs to close."

For reference, I'm on ubuntu 12.04, MATALB 2015a, on a VM (VMWare Workstation).

tkoolen, 2015-07-29:
I can't reproduce this with Matlab 2014a on OSX. Could you start matlab from the command line, get it to crash, and post what's printed to the command line as it's crashing?

aespielberg, 2015-07-30:
Okay, I will do this tonight.

-Andy S.

aespielberg, 2015-07-30:
All that prints to the terminal is the startup message:

MATLAB is selecting SOFTWARE OPENGL rendering.

Incidentally, it did work once just now, but restarting matlab caused it to
no longer work, although I'm not sure why.

-Andy S.

aespielberg, 2015-07-30:
The bug apparently is in the p = pendulumPlant(); line.  Stupidly, I just
assumed it was the dynamicsGradient call

tkoolen, 2015-07-30:
I happened to upgrade to Matlab 2015a today and was reading through the 2014b [release notes](http://www.mathworks.com/help/matlab/release-notes.html#R2014b). Among them was this item:
> New startup switch to opt out of automatically switching to software OpenGL
If MATLAB detects it is running with a graphics driver that has known issues, it automatically switches to software OpenGL®. You can opt out of this behavior by calling MATLAB with the -nosoftwareopengl startup option. See matlab (Windows) or matlab (UNIX).
To re-enable automatic switching to software OpenGL, call MATLAB with the -softwareopengl startup option.
These startup options are not available on Mac platforms.

I'm thinking that there may be issues with graphics since you're running in a virtual machine. Could you try running Matlab from the command line with -nosoftwareopengl?

aespielberg, 2015-07-30:
Still broke it.  :-/.  Probably not a big deal now that I realize this is
with a class I don't care about, but probably good to figure out for future
users.

RussTedrake, 2015-08-27:
aha.  i bet this is the same as #1276 .

closed 2015-08-27 by RussTedrake