RobotLocomotion/drake #1203: MATLAB suddenly very slow - Drake or something else?

aespielberg, 2015-07-28:
Hi,

Optimization runs had been reasonable in speed in the past, but today it has been very slow.  As I am running a linux VM which is running MATLAB which is running drake libraries, there are a lot of potential causes.

I am simply wondering if anything in drake may suddenly cause it to suddenly run more sluggish than usual (from a clean start of MATLAB, no code run yet in session), potentially in cached files that drake creates?  Otherwise, it is likely something else in my chain.

tkoolen, 2015-07-29:
If matlab itself is slow before running any drake code, I would expect it to not be related to drake. Is there anything in your startup.m related to drake?

aespielberg, 2015-07-30:
Yeah, I addpath podst and addpath drake in my startup.m, as well as the
build folder.  But that shouldn't slow it down once it's done with startup,
right?

-Andy S.

tkoolen, 2015-07-30:
It shouldn't, but that's the only thing I can think of that's related to drake. Is matlab still slow when you take those lines out?

aespielberg, 2015-07-30:
The bug apparently is in the p = pendulumPlant(); line.  Stupidly, I just
assumed it was the dynamicsGradient call

aespielberg, 2015-07-30:
It seems fairly transient.  I will check next time it seems to be acting
slowly.

-Andy S.

aespielberg, 2015-08-02:
Can't seem to reproduce.  Closing for now.

closed 2015-08-02 by aespielberg