RobotLocomotion/drake #1229: Consider using TravisCI as an additional test system

rdeits, 2015-08-05:
If we're going to make Drake work well without Matlab, then we're going to need a non-matlab build environment to test with. While I'm sure we could just set up yet another build server, I'd suggest we consider using  a free service like Travis https://travis-ci.org/ to handle the build and test process for us. I've been using it with IRIS and other projects, and I've found it really easy to work with. 

Here are a few advantages I see:
* It automatically integrates with github pull requests to show you the status of the build and test for each PR. For example: https://github.com/rdeits/iris-distro/pull/21 (and it knows to re-run the tests if any new commits are added to the PR). 
* It also doesn't spam the github comments with build output :wink: 
* The entire build server is spun up as a new virtual machine for every build, so there's no weird state persistence (although it's also possible to cache slow parts of the build if you want)
* It's free (for public repos) and open source (but you can also pay them for private repos)
* It allows multiple build configurations. For example, I've set up IRIS to test every commit with both `gcc` and `clang` as separate builds. 

And some disadvantages:
* It doesn't give a cdash-like display of the result of every test separately. Instead, it just reports success or failure for the whole build process. That means that if we want to handle "known failures" in the c++ tests, we have to write that logic ourselves. 
* It's another thing to understand and configure. 

Fortunately, I think the configuration is pretty easy. I'm working on getting drake to build on Travis right now from my fork, and the config script is just:

```yaml
language: cpp

sudo: required

notification:
  email: false

compiler:
  - gcc
  - clang

addons:
  apt:
    sources:
      - ubuntu-toolchain-r-test
    packages:
      - gcc-4.8
      - g++-4.8

install:
  - if [ "$CXX" = "g++" ]; then export CXX="g++-4.8" CC="gcc-4.8"; fi

before_install:
  - make download-all
  - sudo ./install_prereqs.sh ubuntu

script:
  - mkdir build
  - make
  - CTEST_OUTPUT_ON_FAILURE=1 make test
```

rdeits, 2015-08-05:
Oh, and I forgot to mention: the actual primary advantage is that *they* maintain the servers for us. So it should "just work"

rdeits, 2015-08-05:
You can see the build status of my drake fork here: https://travis-ci.org/rdeits/drake

Currently it's failing because there are some actual errors in our build system when matlab is not found. I'm working on fixing those now.

RussTedrake, 2015-08-05:
I'm all for it.  it also would probably not be hard to use their example for the way that our build servers report status.  Looks like there is support in the github api, eg. https://github.com/blog/1227-commit-status-api

rdeits, 2015-08-05:
Huh, cool. It would be really really nice to have the pull request status
attached as metadata instead of sprinkled throughout the comments.

PR #1231 (2015-08-11): Enable Travis CI to build and test our non-matlab components
This branch creates the `.travis.yml` file, which contains the configuration options to build and test Drake with Travis: https://travis-ci.org/

I've also fixed a few build issues that came up when building without matlab installed. We can now run 11 thrilling c++-only unit tests :wink: 

As a demo, I've set up the Travis build to run 4 sets of tests:
* gcc + Release
* gcc + Debug
* clang + Release
* clang + Debug

We can reconfigure the set of tests that get run however we want. 

I've got Travis building on my drake fork right now [... cut]
closed 2015-08-11 by RussTedrake