RobotLocomotion/drake #1230: Problems with install_prereqs.sh

rdeits, 2015-08-05:
I've found a couple of problems with install_prereqs.sh:

1. Since it invokes `apt-get install`, it must be run as root (on ubuntu). But it also calls `make download-all`, which results in all of the external source folders being owned by root (which is bad).
1. Because it calls the Makefile, it invokes cmake and ends up setting up some initial Cmake cache variables. That's a problem because it always sets the BUILD_TYPE to "Release", which makes a debug build impossible.

rdeits, 2015-08-05:
Here are some options:

1. don't call `make download-all` inside `install_prereqs.sh`, just tell users to do so beforehand. We can also simplify the install_prereqs logic to just look inside the `drake` and `externals` folders. 
2. Change `install_prereqs.sh` to be run as user but to call `sudo apt-get install` when appropriate. But figuring out whether `sudo` is needed for a particular system will be very tricky. 

I've already implemented option 1 in my testing branch.

rdeits, 2015-08-05:
Option 2 also requires that we correctly pass the build type through every time we use the makefile.

RussTedrake, 2015-08-05:
I was thinking about this -- it's the reason that the logic is not inside cmake in the first place.  But slipped in calling make download-all.  My bad.  Option 1 seems reasonable.

rdeits, 2015-08-08:
Ok, sounds good. This will be fixed in #1231

PR #1231 (2015-08-11): Enable Travis CI to build and test our non-matlab components
This branch creates the `.travis.yml` file, which contains the configuration options to build and test Drake with Travis: https://travis-ci.org/

I've also fixed a few build issues that came up when building without matlab installed. We can now run 11 thrilling c++-only unit tests :wink: 

As a demo, I've set up the Travis build to run 4 sets of tests:
* gcc + Release
* gcc + Debug
* clang + Release
* clang + Debug

We can reconfigure the set of tests that get run however we want. 

I've got Travis building on my drake fork right now [... cut]
closed 2015-08-11 by RussTedrake