RobotLocomotion/drake #1237: Multiple Matlab pods collections on the same machine will conflict

rdeits, 2015-08-07:
(duplicated from https://github.com/RobotLocomotion/cmake/issues/16 because I don't think most of us in the lab are following the cmake pod issues list)

I recently switched my version of IRIS to use the matlab_pods.cmake functions. Unfortunately, calling pods_configure_matlab_paths() in the IRIS build process breaks my drake-distro installation. My setup is something like this:

```
--- ~
    |--- iris-distro
    |--- drake-distro
```

so I have `~/iris-distro/build/matlab` and `~/drake-distro/build/matlab` on my Matlab path. Unfortunately, this doesn't work, because Drake internally uses `pods_get_base_path()` and similar functions, which it ends up picking up from IRIS instead. This results in weird errors like this:

```
>> addpath_drake
Warning: Invalid file or directory '/Users/rdeits/iris-distro/build/share/java/drake.jar'.
```

(it's looking for `drake.jar` inside IRIS because the IRIS copy of `pods_get_base_path()` is found first). 

Obviously, I could fix this particular problem by putting Drake first on the matlab path, but that will just cause other problems later on. 

There are a few things we could consider doing to fix this:

1. Rename `pods_get_base_path()` and similar functions to include the name of the project (like `pods_get_drake_base_path()`. This would solve the immediate problem, but would cause other, even more obscure, problems if you were to try to maintain two versions of drake on the same machine.
2. Use environment variables to control the matlab path, the same way we do with PATH, PYTHONPATH, LD_LIBRARY_PATH, etc. In this scenario, the `pods_setup_all.sh` function would activate the relevant matlab paths at the same time as it sets up all the other environment variables. On mac and linux, we can just set the MATLABPATH environment variable, but on Windows it looks like we'd have to write this functionality ourselves. 
3. Move the `pods_get_base_path()` functions into the source matlab folder, rather than installing them, so that they're only accessible after calling `addpath_foo()` and thus won't conflict. Of course, once you've called `addpath_foo` and `addpath_bar` then whether you get `foo`'s copy of `pods_get_base_path()` or `bar`'s copy is anybody's guess. 

I personally think that #2 is a nice solution, but we'd also need to combine it with #1 to avoid namespace conflicts. The workflow would be:

* build drake
* don't add anything to the matlab saved path
* instead, when you want to use a particular drake distro, just source the pods_setup_all.sh script in that distro before starting matlab. If you only have one distro, then you can just put that source command in your `.bashrc`. 

This provide a consistent workflow no matter whether you're using the Drake c++ libraries, python tools, or matlab scripts. 

Of course, this also has a number of disadvantages:

* it doesn't work at all on windows unless we write all the code ourselves
* it doesn't work if you launch matlab by double-clicking on the application instead of launching it from a terminal
* it requires our users to change their workflow

I think that the minimal change that would solve the immediate problem is to just do #1. Users with multiple drake installations will still have to manually manage their path within matlab, but they're presumably used to that.

RussTedrake, 2015-08-07:
> (duplicated from RobotLocomotion/cmake#16 because I don't think most of us in the lab are following the cmake pod issues list)

I think you shouldn't have both of those in your path (user error :), unless they are sharing a build directory.  In my mind this is no different than having multiple copies of drake.

rdeits, 2015-08-07:
But what about someone developing their own matlab pod tool? Unless they decide to install into drake/build as well, they'll end up with the same weird failures that I saw (or other failures when their code goes looking for their data files inside drake). 

It makes sense that we would want to keep `pods_get_base_path()` generic for reusability, but the path conflict means that it's hard to reuse them at all. All we have to do to fix it is to add a namespace based on the project name to those scripts (after all, "Namespaces are one honking great idea" :wink: )

Of course, we can always just manipulate the path from within matlab. But I've found that to be remarkably tedious because, for example, my `startup.m` calls `tbxmanager restorepath` to add all of my non-drake matlab toolboxes. So, to edit the matlab path, I: 
* load matlab
* remove all the new path entries which have shown up since I last saved the path
* make the desired change
* save the path

None of that is particularly hard, but it's just additional friction against getting real work done.

RussTedrake, 2015-08-07:
I think you would install it alongside drake, and you did, but make a build directory in that root and let them install to the same place?

rdeits, 2015-08-07:
Yup, that works.

closed 2015-08-10 by RussTedrake