RobotLocomotion/drake #1239: Calling checkDependency('mosek') destroys existing yalmip variables exactly once on drake006

rdeits, 2015-08-07:
????

```
                             < M A T L A B (R) >
                   Copyright 1984-2012 The MathWorks, Inc.
                     R2012b (8.0.0.783) 64-bit (glnxa64)
                               August 22, 2012

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

>> addpath_drake
Adding /tmp/drake-1223-22946/drake/../build/matlab to the matlab path
>> checkDependency('yalmip')
 Calling addpath_yalmip

ans =

     1

>> x = sdpvar(1)
Linear scalar (real, 1 variable)
>> yalmip('nvars')

ans =

     1

>> checkDependency('mosek')
 Calling addpath_mosek

MOSEK Version 7.1.0.33 (Build date: 2015-8-5 05:36:11)
Copyright (c) 1998-2015 MOSEK ApS, Denmark. WWW: http://mosek.com
Platform: Linux/64-X86

For help type: help mosekopt.

ans =

     1

>> yalmip('nvars')

ans =

     0

>> x
Index exceeds matrix dimensions.

Error in sdpvar/depends (line 14)
if ~any(variabletype(x.lmi_variables))

Error in sdpvar/display (line 10)
            vars = depends(X);
```

uh...I'll investigate.

rdeits, 2015-08-07:
Ah, I see. 

The problem is:

* mosek Fusion requires a java library
* we can't modify the java path if there are java variables defined, so we add that library with `javaaddpathProtectGlobals`, which also backs up all global variables to a local variable and the restores them
* but yalmip's variables are stored in global state, and they don't get restored properly for some reason

That's why it only happens once per matlab session (since we only modify the java path once).

rdeits, 2015-08-07:
Actually, it looks like yalmip's state is marked `persistent`, so it's probably getting cleared but not restored. That's...awkward. 

A few things we could do:
* not support or install the mosek java library. I only brought it in for IRIS, and it's no longer used there. 
* try to figure out how to also restore persistent variables after doing a javaaddpathProtectGlobals. Some quick googling didn't show any way to do that, but it may be possible
* just move all the checkDependency('mosek') calls earlier in their respective files. This is such a dumb hack but it would solve the immediate problem. 
* something more radical

In the "more radical" and "probably more effort than it's worth" categories, I feel like we might reconsider the way we do dependencies. I don't like that `checkDependency`, whose name suggests that all it's doing is looking for a program, actually does lots of scary things to the Matlab global state (modifying the path, deleting variables, invalidating your java objects, etc.). Instead, I'd suggest that we use Matlab's packages to handle our namespaces and then install all of our software as Matlab packages into the `build/matlab` folder, along with the relevant java files (or, add the `src/matlab` folder directly to the matlab path to prevent users from having to rebuild after just changing matlab code). Since all matlab packages and java libraries would be present at startup, there would be no issues of global state.

RussTedrake, 2015-08-07:
I really really don't think we should install matlab files.  That would make development very tough; we'd have two copies everywhere and would have to install to test (totally defeating the purpose of an interpreted language)

rdeits, 2015-08-08:
Sure, I totally understand (and agree). 

But that's actually a tangential issue (and it's my bad for mixing it up with this). What we could instead do is package the matlab files up and then have our users add the `src/matlab` (or whatever folder contains `+drake`) to their matlab path permanently. So, there's no `addpath_` scripts, just `drake.examples.Atlas.runWalking` or `import drake.examples.Atlas.*; runWalking()`.

rdeits, 2015-08-08:
But anyway, short term, I guess I'll just move the checkDependency('mosek') calls earlier in the file.

RussTedrake, 2015-08-08:
I do like that idea very much, but you tried packaging everything in drake before and decided it matlab’s packaging support wasn’t quite sufficient… right?  

I agree that checkDependency could get refactored (carefully) to a better name.  maybe even addDependency() .  most of the times it gets called is in fact just to check, but agreed that it is misleading since it changes things.

rdeits, 2015-08-08:
The only problem with packages is that they break the

```
cd examples
runWalking
```

workflow. Otherwise they're great.

But we'd have to decide whether that kind of workflow change is worthwhile.

RussTedrake, 2015-08-08:
And also in the eg systems directories, it's gross to have to import the current directory to call your neighboring functions.

But if it makes it easier / better to use drake in other projects, then we should still consider it.

rdeits, 2015-08-13:
I do think we should seriously consider it, even though it breaks an existing workflow. Global state is bad. Manipulation of the path variable creates hidden global state and can cause arbitrary scary things to happen. 

Also, it's possible that if we move to a package system, we might be able to do a lot more of our tests within the same matlab session. We switched to re-launching matlab for every test because things like the java path manipulation could cause bad interactions between tests. If we agree never to manipulate the path variables, then we can probably run a lot more tests safely side by side. I think that could actually be a huge advantage. Our build servers spend a *ton* of time restarting matlab, re-JIT-ing the files, and shutting down Matlab. If we can trust the tests to be independent, we could eliminate that waste.

RussTedrake, 2015-08-13:
It feels like we’re far away from the point where i would have that level of trust..

RussTedrake, 2016-05-01:
renamed this to reflect the discussion that resulted.  but resolving this as "won't fix" for now.

closed 2016-05-01 by RussTedrake