RobotLocomotion/drake #1240: Consider not randomizing test input

rdeits, 2015-08-08:
I think it might be worth having a discussion about whether the amount of randomization we have in our unit tests is actually a good thing. 

My understanding of the rationale for randomized tests is that it lets us test a much broader input set than  we could normally cover by only exploring a tiny part of that input set each time the build runs. And that certainly has its benefits. The fact that we record the RNG seed for every test is also awesome, and it's a huge boost to our ability to understand problems. 

But it also has two pretty serious consequences:
* it makes it very hard to tell which commit was responsible for breaking the build, since failures can (and do) happen at random
* random failures can act as a mask for new problems that creep in

I actually think that second point is more troubling. Consider an example (loosely based on something that happened today):

I have a test case which generates an environment, plans some footsteps, and then executes a walking plan. It's been randomly failing occasionally since DRC, so it's marked as a known issue. But it turns out that somewhere along the line something else broke, and now even if that random failure does not occur, the test will just crash later with a frame mismatch error. But we never noticed because the random failure was already masking the status of that test. Essentially, if any test is randomly failing, then from our perspective it's indistinguishable from *consistently* failing. 

So, as we start to move into c++ (and thus have more computational capacity to test lots of input), I think it might be worth considering not randomizing our test input. Or, to be more specific, when we generate tests that use random data, that we make sure that the RNG seed is the *same* every time. We won't cover as much of the input space, but at least we'll have an easier time identifying bad changes. 

What do other people think?

hongkai-dai, 2015-08-08:
I think we need the randomness in the test. For test-driven programming, we need extensive tests for the corner cases, and we do not usually cover all the corner cases intentionally. Randomness is helpful for covering more test cases. And I think many of our tests are for small piece of code. It is easy to isolate which commit breaks the test for that small piece of code.

What about a test with no randomness as the baseline (so we will understand some commit breaks the test), and a nightly build that has random seed? This will double the test time. 

Another approach is that we can also create a "norandom" tag (like “notest") in the test files that can be identified by unitTest.m, that for test on small piece of code, it does not need such  "norandom" tag, but for some code that we want a no-random test, we add this tag, and the test run for twice. The first time without randomness, and the second time with randomness.

RussTedrake, 2015-08-09:
I’m torn on this one.  I definitely see the value in having consistent unit tests, but also in having more strenous unit tests with randomness.  

While it’s true that our knownIssue logic doesn’t distinguish between continual failure and rare failure, it *is* possible to write pretty specific error checking logic.  You can string parse the specific console output, the error name, etc.  So I think that the short term focus should be on writing more specific known issue logic.

tkoolen, 2015-08-11:
I also think it's a difficult decision, but I lean towards not having randomness.

As the number of test cases grows, a small probability of any given test failing randomly will mean that it's more and more likely for at least one to fail on every build, causing the build servers to turn into boys who cry wolf. The argument that having randomness can help find bugs is only valid if people still care enough to look at a failing test case and fix the bug.

By the way, it is worth noting though that using the same random seed is *not* guaranteed to give the same result across versions of Matlab or across architectures (found this out the hard way).

avalenzu, 2015-08-11:
It seems like we're using our unit tests in two different ways here:

1. Checking pull-requests for regressions (randomness hurts us here)
2. Checking master for corner-cases that we've missed (randomness is essential here)

One solution (similar to @hongkai-dai's suggestion) would be to look for a CONTINUOUS_TESTING_ONLY flag in the text of the file. Tests containing this flag would only be run in the nightly tests on master. Then we could make it a policy to only allow randomness in files with the CONTINUOUS_TESTING_ONLY flag, which would give a us a set of deterministic tests for evaluating pull-requests.

RussTedrake, 2015-08-11:
How about this:

I can make the randomness yet another cmake option.  Then the pull requests will turn it off.  I can have it on for the nightly builds, and can open issues as necessary.

psiorx, 2015-08-11:
I agree with @avalenzu that we're supporting two different scenarios here but it would be a pain to have to manually annotate test cases.  This also doesn't address the problem of a single test providing utility for both scenarios.  It forces a test to fall in one of the buckets.

I'm in favor of a more global option (potentially cmake) that could be toggled for different testing scenarios.

tkoolen, 2015-08-11:
I think @avalenzu's split between pull requests and master is a great idea. Maybe we could even have the test runner set a fixed random seed or call `rng('shuffle')` before a test based on some flag in the text of the file.

tkoolen, 2015-08-11:
(that's for Matlab only of course). Come to think of it, there wouldn't really be a need for the flag in the text of the file anymore.

RussTedrake, 2015-08-11:
As @psiorx says -- (mostly) disabling/enabling the randomness globally is easy, but adding flags to all the relevant tests is not.   Maintaining which files are meant for continuous testing vs nightly testing might be a pain, too -- unless we have some policy for it (eg example dirs are only run nightly -- though i don't like that particular one).

I still like the random seed disabled for pulls and enabled for nightly idea best

rdeits, 2015-08-13:
I also like the idea of separating pull request builds from nightly builds, since they do serve different purposes. Fixing the seed for pull requests seems like a pretty good solution all around.

PR #1258 (2015-08-15): unit test randomization is now a cmake option (disabled by default)
resolves #1240 .
closed 2015-08-15 by RussTedrake