RobotLocomotion/drake #1255: runAtlasWalkingPlanning fails

rdeits, 2015-08-13:
```
>> runAtlasWalkingPlanning
Error using CoordinateFrame/findTransform (line 291)
Could not find any transform between atlasFrames.AtlasState and
atlasPosition

Error in PPTrajectory/inFrame (line 90)
        tf =
        findTransform(obj.getOutputFrame,frame,struct('throw_error_if_fail',true));
        
Error in PPTrajectory/subsref (line 503)
        [varargout{:}] = builtin('subsref',a,s);

Error in Visualizer/playback (line 85)
        xtraj = xtraj.inFrame(obj.getInputFrame);  % try to convert
        it

Error in runAtlasWalkingPlanning (line 83)
v.playback(xtraj, struct('slider', true));
```

This is also causing all the walking planning demos to fail on my machine. I'll look into it.

psiorx, 2015-08-13:
This also happens for the running planner.  @tkoolen was looking at this yesterday.

rdeits, 2015-08-13:
I think the build servers aren't catching it because they have visualizers turned off.

RussTedrake, 2015-08-13:
I seem to remember addressing something like this in 35dbcd31f1c58247ee649dc9d2e5abbd55cac2e4

rdeits, 2015-08-13:
Reverting 35dbcd31f1c58247ee649dc9d2e5abbd55cac2e4 fixes this particular issue for me.

tkoolen, 2015-08-13:
As for `testRunningPlanner`. The fact that it doesn't work anymore was completely my own fault. When I got rid of the old kinematics, I made the doKinematics call in SimpleDynamicsFullKinematicsPlanner use mex because I thought the only thing preventing this before was the need for gradients of the CMM. But the more important thing is that multiple kinsols need to be valid at the same time, while currently calling mex doKinematics overwrites the cache in RigidBodyManipulator. John and I are actually working on getting the kinematics cache out of RBM and using a separate DrakeMexPointer to a KinematicsCache object, to be stored in the Matlab kinsol struct.

tkoolen, 2015-08-13:
`runAtlasRunning` does have frame issues however, possibly unrelated:
```
Error using CoordinateFrame/autoConnect (line
485)
Automatic connection failed.  Could not find any
possible connections between sys1 and sys2

Error in DynamicalSystem/mimoCascade (line 56)
connection =
autoConnect(sys1.getOutputFrame,sys2.getInputFrame,connection);

Error in runAtlasRunning (line 317)
sys = mimoCascade(sys,v,[],[],output_select);
```

closed 2015-08-27 by RussTedrake